jsk-ros-pkg/jsk_visualization #301: オーバーレイのプロットが重い

yuki-asano, 2015-01-30:
２００個表示すると10fpsくらいになる

garaemon, 2015-01-30:
https://github.com/jsk-ros-pkg/jsk_visualization/pull/303
これで解決されるかも

garaemon, 2015-01-30:
https://github.com/jsk-ros-pkg/jsk_visualization/pull/304
これでどうだろう

garaemon, 2015-01-30:
https://github.com/jsk-ros-pkg/jsk_visualization/pull/305
これでどうだろうか

garaemon, 2015-01-31:
200個はかなり厳しいようですね。VRAMとのやり取りに時間がかかっています

garaemon, 2015-01-31:
ちなみに、サイズを小さくする(例えば128x128を10x10にする)と速くなります

yuki-asano, 2015-01-31:
やっぱり3fpsですね。
outputのグラフが消えてしまいました。
tensionもしきい値が低いのかすぐに赤くなってしまいます。

garaemon, 2015-01-31:
outputってなんですか

yuki-asano, 2015-01-31:
アクチュエータの出力です。
一つのアクチュエータにつき二つ出ていたグラフの片方になります。

garaemon, 2015-01-31:
どっち側ですか?

yuki-asano, 2015-01-31:
左右で言う所の右側です

garaemon, 2015-01-31:
円グラフか折れ線グラフかで教えてください

yuki-asano, 2015-01-31:
そういう意味でしたか。
円グラフです。

garaemon, 2015-01-31:
確認します

garaemon, 2015-01-31:
> tensionもしきい値が低いのかすぐに赤くなってしまいます。

何割くらいから赤くなってほしいですか?
今は線形に色を変化させてしまっているのでいまいちかもしれない、非線形にしようかと思います

yuki-asano, 2015-01-31:
今の設定がどのくらいか分からないのですが、
確か昔レンジが広すぎるとよくない感じだった気がするので、
30kgf(300N)を上限として、
20kgf(200N)くらいから赤くなるとうれしいのかなと思います。

garaemon, 2015-01-31:
piechart消える問題は解決されたかと思います、チェックしてくれると助かります

yuki-asano, 2015-01-31:
piechart復活しました。
ありがとうございます。

garaemon, 2015-01-31:
大元の高速化については、ロボットモデルのみで20fps以上がでるようになったら最適化しようと思います。

この最適化は結構難しいです。。。

yuki-asano, 2015-01-31:
はい、ロボットモデルの削減もおいおいやっていきたいと思います。

garaemon, 2015-01-31:
その前に緊張やこびあんの高速化ですね

garaemon, 2015-01-31:
一応今すぐできる所までは高速化したのでclose

closed 2015-01-31 by garaemon