jsk-ros-pkg/jsk_visualization #401: How can I use rqt_service_buttons?

wkentaro, 2015-05-20:
by running `rosrun rqt_gui rqt_gui` there are no buttons.
I think it should load default layout yaml file..
[image]

In 553e50bf742ecdc73a7d05b449f8fb99947d5317 ,
`rosrun jsk_rqt_plugins rqt_service_buttons` works.
[image]

wkentaro, 2015-05-20:
Is this bug? Or should I do something more?

garaemon, 2015-05-20:
You can find gear button on the right top

wkentaro, 2015-05-20:
Thanks. It was hard to find the way to use this.
because below does not work.
```
rosrun jsk_rqt_plugins rqt_service_buttons _layout_file:=/jsk_rqt_plugins/resource/service_button_layout.yaml
```

[image]

wkentaro, 2015-05-20:
I had to type full path after click the gear mark.
To make it easy to use this package for other users, I will add README for this.

closed 2015-05-20 by wkentaro

garaemon, 2015-05-20:
jsk_visualization is not well documented, please add README

wkentaro, 2015-05-21:
I sent PR : https://github.com/jsk-ros-pkg/jsk_visualization/pull/404/files