jsk-ros-pkg/jsk_visualization #491: How to use video_capture?

wkentaro, 2015-10-02:
[image]

I have StatueError (NOT Writable)

garaemon, 2015-10-02:
I found that it is difficult to know a file is writable without creating file.
Now the status error is based on `access` result but it returns true if **file exists** and file is writable.

wkentaro, 2015-10-02:
so should the user do `touch output.avi` at `~/.ros`?

garaemon, 2015-10-02:
> so should the user do touch output.avi at ~/.ros?

yes but I disabled this feature. https://github.com/jsk-ros-pkg/jsk_visualization/pull/492

wkentaro, 2015-10-02:
> I found that it is difficult to know a file is writable without creating file.

Is that impossible to check the access right of the directory in which the movie file will be created?

garaemon, 2015-10-02:
> Is that impossible to check the access right of the directory in which the movie file will be created?

possible, just a second.

garaemon, 2015-10-02:
done https://github.com/jsk-ros-pkg/jsk_visualization/pull/494

garaemon, 2015-10-02:
In fact, I don't recomment to use video capture because it cannot resolve fps correctly.
 I recommend to use glc.

closed 2015-10-02 by garaemon