jsk-ros-pkg/jsk_visualization #500: [jsk_rviz_plugins] Can we rotate image with OverlayImag or Camera?

wkentaro, 2015-10-05:
suggested by @eisoku9618 

Sometimes, camera image is rotated with fixed angle. (that is not bug of software but default camera pose attached to robot)
In that case, the user want to rotate the image with the fixed angle on rviz.

Any idea about this?

garaemon, 2015-10-05:
http://wiki.ros.org/image_rotate

garaemon, 2015-10-05:
`jsk_pcl_ros/ImageRotate` is same to official image_rotate.
historical reason behind it.
https://github.com/jsk-ros-pkg/jsk_recognition/issues/32
https://github.com/ros-perception/image_pipeline/pull/62
https://github.com/ros-perception/image_pipeline/pull/63

k-okada, 2015-10-05:
https://github.com/ros-perception/image_pipeline/tree/indigo/image_rotate
or
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_perception/src/virtual_camera_mono.cpp

closed 2015-10-16 by garaemon