jsk-ros-pkg/jsk_visualization #502: tile images on top page on readthedocs

wkentaro, 2015-10-05:
like: http://matplotlib.org/gallery.html
suggested at: https://github.com/jsk-ros-pkg/jsk_visualization/issues/485#issuecomment-145431501

wkentaro, 2015-10-05:
方法案

* tile_images.py(pure python without opencv)をdoc/以下に置く
* tile_images.pyはドキュメント生成時に実行される
* tile_images.pyはdoc/jsk_rviz_plugins/imagesを見てすべてをタイル状にした画像を出力する
* doc/jsk_rviz_plugins/index.rstにタイル状の画像を載せる

wkentaro, 2015-10-05:
ただこうなると、それぞれの画像にリンク貼りたいとかになりますよね。。
ということは、index.rstを生成する実行ファイルという話になります。

garaemon, 2015-10-05:
まじめにやると、mdやらrstのASTを作って画像のリストを作るみたいに成りますが、sed的なもので適当にできるんじゃないかなと思ってます。

garaemon, 2015-10-05:
たとえば、texから画像ファイルを抜き出して、そのファイルを調べるスクリプト

```
#!/bin/bash

TEXFILE=2015-Humanoids-ueda-drc-system.tex

FILES=$(grep "\.png\|\.jpg\|\.jpeg\|\.eps\|\.PNG\|\.JPG\|\.JPEG\|\.EPS" $TEXFILE | sed -e "s/.*{\(.*\)}.*/\1/")
ls -lht $FILES -S
```

wkentaro, 2015-10-05:
画像ファイルはすべてimagesフォルダにあるからそれを全部並べればいいということではないですか？

garaemon, 2015-10-05:
そうかも。png -- mdの対応関係がほしいと思うので、適当に正規表現で抜き出したら対応付けは作れる気がしますね

wkentaro, 2015-10-05:
確かに画像ファイルの名前が必ずしも
* mask_image_to_point_indices1.png
* mask_image_to_point_indices2.png
* mask_image_to_point_indices.md
みたいにはなっていなそうですね。

garaemon, 2015-10-05:
逆にそういう命名速に合わせる、というのもありですね。

wkentaro, 2015-10-05:
個人的には命名則派ですかね。シンプルなので。

wkentaro, 2015-10-05:
rstでtableを使えばこんな感じで出来ました。
https://github.com/wkentaro/inbox/blob/master/rst-tile-images-with-link/index.rst

wkentaro, 2015-10-05:
これならスクリプトで作るのも簡単そうです。

wkentaro, 2015-10-06:
こんな感じでしょうか。
[image]

garaemon, 2015-10-06:
良い感じだと思います

commit 776008f4 (2015-10-06): Add image gallery to index of document

Closes #502
closed 2015-10-06 by garaemon