jsk-ros-pkg/jsk_visualization #513: [jsk_rviz_plugins] Cannnot build jsk_rviz_plugins because of template arguments error

orikuma, 2015-10-16:
jsk_rviz_plugins cannot built in jaxonred (dependent package of jsk_footstep_controller).
- pcl_ros and pcl_conversions are built from source (hydro-devel)
- jsk_recognition and jsk_visualization are HEAD of master 

```
[jsk_rviz_plugins] ==> '/home/leus/ros/hydro/build/jsk_rviz_plugins/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/leus/ros/hydro/build/jsk_rviz_plugins'
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_py
[  0%] [  0%] Built target geometry_msgs_generate_messages_cpp
Built target geometry_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_lisp
[  7%] [  7%] Built target std_msgs_generate_messages_cpp
Built target jsk_rviz_plugins_generate_messages_eus
[ 15%] [ 22%] Built target jsk_rviz_plugins_generate_messages_py
[ 28%] Built target jsk_rviz_plugins_generate_messages_cpp
Built target jsk_rviz_plugins_generate_messages_lisp
[ 28%] [ 28%] Built target jsk_rviz_plugins_generate_messages
Built target jsk_rviz_plugins_gencpp
[ 28%] [ 29%] [ 30%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/polygon_array_display.cpp.o
[ 30%] [ 31%] [ 31%] [ 31%] [ 32%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_record_action.cxx.o
Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_polygon_array_display.cxx.o
Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_overlay_text_display.cxx.o
Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_cancel_action.cxx.o
Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_overlay_menu_display.cxx.o
Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_overlay_image_display.cxx.o
Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_normal_display.cxx.o
[ 33%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_overlay_camera_display.cxx.o
[ 34%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_plotter_2d_display.cxx.o
[ 34%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_pie_chart_display.cxx.o
[ 35%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_diagnostics_display.cxx.o
[ 35%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_quiet_interactive_marker_display.cxx.o
[ 36%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_bounding_box_array_display.cxx.o
[ 37%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_torus_array_display.cxx.o
[ 37%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_overlay_diagnostic_display.cxx.o
[ 38%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_target_visualizer_display.cxx.o
[ 39%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_simple_occupancy_grid_array_display.cxx.o
[ 39%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_camera_info_display.cxx.o
In file included from /usr/include/boost/signals2.hpp:20:0,
                 from /usr/include/pcl-1.7/pcl/io/boost.h:73,
                 from /usr/include/pcl-1.7/pcl/io/file_io.h:43,
                 from /usr/include/pcl-1.7/pcl/io/pcd_io.h:44,
                 from /home/leus/ros/hydro/src/pcl_conversions/include/pcl_conversions/pcl_conversions.h:70,
                 from /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_recognition_utils/include/jsk_recognition_utils/pcl_conversion_util.h:40,
                 from /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_recognition_utils/include/jsk_recognition_utils/geo/polygon.h:50,
                 from /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_recognition_utils/include/jsk_recognition_utils/geo_util.h:74,
                 from /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/include/jsk_pcl_ros/geo_util.h:39,
                 from /home/leus/ros/hydro/src/jsk-ros-pkg/jsk_visualization/jsk_rviz_plugins/src/polygon_array_display.cpp:40:
/usr/include/boost/signals2/signal_type.hpp:78:9: : wrong number of template arguments (6, should be 5)
/usr/include/boost/parameter/parameters.hpp:714:8: : provided for template<class PS0, class PS1, class PS2, class PS3, class PS4> struct boost::parameter::parameters
/usr/include/boost/signals2/signal_type.hpp:83:9: : parameter_spec is not a class or namespace
/usr/include/boost/signals2/signal_type.hpp:83:29: : expected unqualified-id before < token
/usr/include/boost/signals2/signal_type.hpp:99:11: : args was not declared in this scope
/usr/include/boost/signals2/signal_type.hpp:105:9: : template argument 1 is invalid
/usr/include/boost/signals2/signal_type.hpp:108:31: : args was not declared in this scope
/usr/include/boost/signals2/signal_type.hpp:108:67: : template argument 1 is invalid
/usr/include/boost/signals2/signal_type.hpp:111:31: : args was not declared in this scope
/usr/include/boost/signals2/signal_type.hpp:111:94: : template argument 1 is invalid
/usr/include/boost/signals2/signal_type.hpp:115:31: : args was not declared in this scope
/usr/include/boost/signals2/signal_type.hpp:115:97: : template argument 1 is invalid
/usr/include/boost/signals2/signal_type.hpp:121:13: : args was not declared in this scope
/usr/include/boost/signals2/signal_type.hpp:124:11: : template argument 1 is invalid
/usr/include/boost/signals2/signal_type.hpp:128:31: : args was not declared in this scope
/usr/include/boost/signals2/signal_type.hpp:128:69: : template argument 1 is invalid
[ 40%] [ 41%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_open_all_tool.cxx.o
Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_close_all_tool.cxx.o
[ 42%] [ 42%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_object_fit_operator.cxx.o
Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_transformable_marker_operator.cxx.o
[ 42%] [ 43%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_robot_command_interface.cxx.o
Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_empty_service_call_interface.cxx.o
[ 44%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_pictogram_display.cxx.o
[ 44%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_pictogram_array_display.cxx.o
[ 45%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/view_controller/moc_tablet_view_controller.cxx.o
[ 46%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_tablet_controller_panel.cxx.o
[ 46%] Building CXX object CMakeFiles/jsk_rviz_plugins.dir/src/moc_video_capture_display.cxx.o
make[2]: *** [CMakeFiles/jsk_rviz_plugins.dir/src/polygon_array_display.cpp.o]  1
make[2]: *** ....
make[1]: *** [CMakeFiles/jsk_rviz_plugins.dir/all]  2
make: *** [all]  2
[jsk_rviz_plugins] <== '/home/leus/ros/hydro/build/jsk_rviz_plugins/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' failed with return code '2'
```

garaemon, 2015-10-16:
テスト通っているので環境がなにかおかしいんじゃないかな

#define BOOST_MAXなんとかを7にしないといけない系のエラーだけど

garaemon, 2015-10-16:
jaxonredはjsk_rviz_pluginsが最新じゃなかったです。
pullしたので確認してください

orikuma, 2015-10-16:
確認不足でした. 申し訳ございません.
無事buildが通りました.

closed 2015-10-16 by orikuma

garaemon, 2015-10-16:
https://github.com/jsk-ros-pkg/jsk_common/issues/1187

issue報告用ツールがあるといいですね