jsk-ros-pkg/jsk_visualization #554: [tile_image.py] Highre resolution

garaemon, 2015-12-20:
I need more higher resolution image outputted by tile_image.py

[image]

sample launch

```xml
<launch>
  <arg name="INPUT" default="/multisense_local/left/image_rect_color" />
  <node pkg="jsk_perception" type="hsv_decomposer" name="hsv">
    <remap from="~input" to="$(arg INPUT)" />
  </node>
  <node pkg="jsk_perception" type="lab_decomposer" name="lab">
    <remap from="~input" to="$(arg INPUT)" />
  </node>
  <node pkg="jsk_perception" type="tile_image.py" name="tile_image">
    <rosparam subst_value="true">                                                                                                               input_topics: [$(arg INPUT),                                                                                                        
                     hsv/output/hue, hsv/output/saturation, hsv/output/value,                                                                                  lab/output/l, lab/output/a, lab/output/b]                                                                            
    </rosparam>
  </node>
  <node pkg="image_view" type="image_view" name="result_view">
    <remap from="image" to="tile_image/output" />
  </node>
</launch>

```

garaemon, 2015-12-20:
My proposal is

* to add column parameter as dynamic_reconfigure (now it is estimated by sqrt)
* to use cv2.vconcat and cv2.hconcat instead of matplotlib concatenation to keep original resolution

wkentaro, 2015-12-20:
The initial column width would be max width of the images?

wkentaro, 2015-12-20:
It's easy to concatenate images if these images size are same, but if not, it's not so simple..

wkentaro, 2015-12-20:
do you imagine like this?
https://github.com/wkentaro/apc-od/pull/20#issuecomment-157343007

I'm using this function to do this.
https://github.com/lisa-lab/pylearn2/blob/master/pylearn2/packaged_dependencies/theano_linear/imaging.py#L191

garaemon, 2015-12-20:
> do you imagine like this?
> wkentaro/apc-od#20 (comment)

Yes,

wkentaro, 2016-04-28:
this is fixed via https://github.com/jsk-ros-pkg/jsk_recognition/pull/1518

closed 2016-05-05 by k-okada