jsk-ros-pkg/jsk_visualization #615: Kinetic version

yukkysaito, 2016-07-28:
Is it possible to make kinetic version?

k-okada, 2016-07-29:
if you install some package from source, I think you can, please find the list of required packges and add to this repository as .travis.rosinstall.kinetic
https://github.com/jsk-ros-pkg/jsk_visualization/commit/d80fac31ffbbaa12d37608170da6f3d836539667

closed 2016-09-20 by yukkysaito

k-okada, 2016-09-21:
thanks to https://github.com/ros-gbp/view_controller_msgs-release/issues/1#issuecomment-247172123, I can release jade version  https://github.com/ros/rosdistro/pull/12781, as for kinetic, we need to wait
dynamic_tf_publisher, image_view2, jsk_pcl_ros, jsk_recognition_msgs, jsk_topic_tools, pr2eus_moveit,  roseus.....