jsk-ros-pkg/jsk_visualization #625: [jsk_rqt_plugins] ColorHistogramのヒストグラムプロットが出来ない(color_histogram_sample.launch)

ishiguroJSK, 2016-08-10:
`roslaunch jsk_perception color_histogram_sample.launch `
がそのままでは失敗します．
まずそもそも，launch内でrqt_histogram_plotに渡しているヒストグラムのトピック名が違っている気がしていて，
そのままだと
```
rqt_plot.rosplot.RosPlotException: Invalid topic spec [/usb_cam/image_raw/hue_histogram/histogram]: 'Image' object has no attribute 'hue_histogram'
Traceback (most recent call last):
  File "/home/ishiguro/catkin_ws/chidori/src/jsk_visualization/jsk_rqt_plugins/src/jsk_rqt_plugins/hist.py", line 150, in update_plot
    data_x, data_y = self._rosdata.next()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_plot/rosplot.py", line 142, in next
    raise self.error
rqt_plot.rosplot.RosPlotException: Invalid topic spec [/usb_cam/image_raw/red_histogram/histogram]: 'Image' object has no attribute 'red_histogram'
Traceback (most recent call last):
  File "/home/ishiguro/catkin_ws/chidori/src/jsk_visualization/jsk_rqt_plugins/src/jsk_rqt_plugins/hist.py", line 150, in update_plot
    data_x, data_y = self._rosdata.next()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_plot/rosplot.py", line 142, in next
    raise self.error
```
みたいなエラーが出るため，color_histogram_sample.launch内の
```
  <node pkg="jsk_rqt_plugins" type="rqt_histogram_plot" name="red_histogram"
        args="/usb_cam/image_raw/red_histogram/histogram"/>
  <node pkg="jsk_rqt_plugins" type="rqt_histogram_plot" name="hue_histogram"
        args="/usb_cam/image_raw/hue_histogram/histogram"/>
</launch>
```
のargs部を
```
 <node pkg="jsk_rqt_plugins" type="rqt_histogram_plot" name="hue_histogram"
        args="/color_histogram/hue_histogram"/>
```
などに書き換えたところ，上記のエラーは止まったものの，image_view2内でマウスによる赤枠選択を行った後に
`[ERROR] [WallTime: 1470812127.683535] Topic/Field name '/color_histogram/hue_histogram' has unsupported '<class 'jsk_recognition_msgs.msg._ColorHistogram.ColorHistogram'>' type.List of float values and jsk_recognition_msgs/HistogramWithRange are supported.`

というエラーが出ます．
ちなみに
`roslaunch jsk_rqt_plugins sample_histogram.launch`
は正常に動作しているので，ヒストグラムのデータ型の問題(
`std_msgs/Float32MultiArray`→◯，
`jsk_recognition_msgs/HistogramWithRange`→◯，
`jsk_recognition_msgs/ColorHistogram`→×
)な気がしています．

wkentaro, 2016-08-10:
```
 <node pkg="jsk_rqt_plugins" type="rqt_histogram_plot" name="hue_histogram"
        args="/color_histogram/hue_histogram/histogram"/>
```
にするとどうでしょう。

ishiguroJSK, 2016-08-10:
できちゃった。

wkentaro, 2016-08-10:
エラーメッセージに

> List of float values and jsk_recognition_msgs/HistogramWithRange are supported.

とあるので、Float型のListになっているjsk_recognition_msgs/ColorHistogramのhistogramフィールドはプロットできるということです。

```
% rosmsg show jsk_recognition_msgs/ColorHistogram  
std_msgs/Header header
  uint32 seq
  time stamp
  string frame_id
float32[] histogram
```

ishiguroJSK, 2016-08-10:
なるほど・・・しかし，どちらもjsk_recognition_msgs/ColorHistogramなのはこういうものなんでしょうか
```
$ rostopic type /color_histogram/hue_histogram
jsk_recognition_msgs/ColorHistogram
$ rostopic type /color_histogram/hue_histogram/histogram
jsk_recognition_msgs/ColorHistogram

$ rostopic echo /color_histogram/hue_histogram/histogram
[3917.0, 7314.0, 721.0, 140.0, 153.0, 183.0, 79.0, 114.0, 140.0, 199.0]
---
$ rostopic echo /color_histogram/hue_histogram
header: 
  seq: 1
  stamp: 
    secs: 1470815748
    nsecs: 753657870
  frame_id: head_camera
histogram: [2224.0, 280.0, 704.0, 1264.0, 2156.0, 596.0, 32.0, 316.0, 92.0, 32.0]
---
```

wkentaro, 2016-08-10:
なるほど、rostopicへの変更が必要そうですね。
https://github.com/ros/ros_comm/blob/kinetic-devel/tools/rostopic/src/rostopic/__init__.py#L912

ishiguroJSK, 2016-08-10:
なにはともあれ，デバッグありがとう．おかげで正常に動作確認出来ました．
次はhttps://github.com/jsk-ros-pkg/jsk_visualization/issues/625のsampleとtestが動作することを確認してdocに追記ですね．
(rostopicの件は正直よく分からんとです・・・)

wkentaro, 2016-08-23:
It seems this issue is resolved. Please close this.

closed 2016-08-23 by k-okada