jsk-ros-pkg/jsk_visualization #629: jsk_rviz_plugins OverlayText - How to show multiple texts, and what fonts are available?

marklewis3000, 2016-08-19:
I have 2 questions about using jsk_rviz_plugins / OverlayText

The OverlayText demo shows the following text being displayed:
    "You can write several text to show the operator  ...   you can choose font"

1) How can you display more than one OverlayText at the same time?
    Hopefully without requiring one publisher for each text, maybe using and ID or Name??

2) What other fonts are available besides "DejaVu Sans Mono" ?

Thank you

wkentaro, 2016-08-20:
> 1) How can you display more than one OverlayText at the same time?
Hopefully without requiring one publisher for each text, maybe using and ID or Name??

To display text more than one OverlayText, the plugin requires one publisher for each text.

wkentaro, 2016-08-20:
Probably #630 works.

wkentaro, 2016-08-23:
Fixed?

marklewis3000, 2016-08-23:
Sorry, but to test this, where is the latest binary/debian or how do I find and install the latest build?

wkentaro, 2016-08-23:
```
cd
mkdir -p catkin_ws/jsk_visualization_ws/src
cd catkin_ws/jsk_visualization_ws/src
git clone https://github.com/jsk-ros-pkg/jsk_visualization.git
cd ..
catkin init
rosdep install --from-path . -r -y
catkin build jsk_rviz_plugins
```

marklewis3000, 2016-08-23:
Sorry. I've been using catkin_make instead of catkin. So I installed catkin and pip, and I have now tried to build with catkin using your instructions above, and I think it completed. How then do you install the newly generated plugin into my rviz?

wkentaro, 2016-08-24:
```
source ~/catkin_ws/jsk_visualization_ws/devel/setup.bash
rospack find jsk_rviz_plugins  # this should be ~/catkin_ws/jsk_visualization_ws/src/jsk_visualization/jsk_rviz_plugins
roslaunch jsk_rviz_plugins overlay_sample.launch
```

marklewis3000, 2016-08-24:
Yes, once I checkbox 'Overtake Color Properties' the 'font' property appears, and I can select from among about 100 listed fonts.

wkentaro, 2016-08-24:
Sounds good. If your problem is resolved, please close this issue.

marklewis3000, 2016-08-24:
Latest update shows the available fonts for OverlayText. Closing issue

closed 2016-08-24 by marklewis3000