jsk-ros-pkg/jsk_visualization #634: caption contents for Plotter2D and PieChart in jsk_rviz_plugins?

marklewis3000, 2016-09-02:
The caption for Plotter2D is simply '**Plotter2D**', and likewise, the caption for PieChart is simply '**PieChart**'.

I would like to customize the caption as needed, such as '**Newtons**' and  '**Meters/Sec**'.

How can this be done? I don't see a property for the contents of the captions.

Thank you

k-okada, 2016-09-12:
Please look at #636, that will solve your problem, Also this will be released as deb very soon.

k-okada, 2016-09-12:
may be we do not have to change anything, just to rename plugins display name as described in http://docs.ros.org/jade/api/rviz/html/user_guide/#naming-displays,

[image]
[image]

Hope this helps

marklewis3000, 2016-09-14:
Using the 'Rename' button near the bottom of the' Displays' panel does change the Caption. Thank you!

closed 2016-09-14 by marklewis3000