jsk-ros-pkg/jsk_visualization #648: How to use the jsk_rviz_plugins for Kinetic?

longjie, 2016-11-28:
Hello,

What is the easiest way to use the jsk_rviz_plugins for Kinetic?
Is there any script files or .rosinstall to get all I need?

Thank you,

wkentaro, 2016-11-28:
Currently, we don't support kinetic and no .rosinstall for that.
At first, we need to create `.travis.kinetic` like [`.travis.hydro`](https://github.com/jsk-ros-pkg/jsk_visualization/blob/master/.travis.rosinstall.hydro).

longjie, 2016-11-28:
I see, it must not be difficult even for me.
I just wanted to know if I have a wiser way.
Thank you!

wkentaro, 2016-11-28:
```
% for pkg in $(rospack depends jsk_rviz_plugins); do                                                                                          │···············
  rosdep resolve $pkg --rosdistro kinetic --os=ubuntu:xenial &>/dev/null || echo $pkg;                                                        │···············
done                                                                                                                                          │···············
jsk_recognition_msgs                                                                                                                          │···············
jsk_recognition_utils

% for pkg in $(rospack depends jsk_rqt_plugins); do
  rosdep resolve $pkg --rosdistro kinetic --os=ubuntu:xenial &>/dev/null || echo $pkg;
done

% for pkg in $(rospack depends jsk_interactive_marker); do
  rosdep resolve $pkg --rosdistro kinetic --os=ubuntu:xenial &>/dev/null || echo $pkg;
done
ml_classifiers
libccd
fcl
jsk_recognition_msgs
jsk_recognition_utils
resized_image_transport
robot_self_filter
jsk_pcl_ros_utils
rethink_ee_description
baxter_description
jsk_pcl_ros
jsk_rviz_plugins
euscollada
pr2_controllers_msgs
pr2eus
pr2eus_moveit

% for pkg in $(rospack depends jsk_interactive); do
  rosdep resolve $pkg --rosdistro kinetic --os=ubuntu:xenial &>/dev/null || echo $pkg;
done
ml_classifiers
libccd
fcl
jsk_recognition_msgs
jsk_recognition_utils
resized_image_transport
robot_self_filter
jsk_pcl_ros_utils
rethink_ee_description
baxter_description
jsk_pcl_ros
jsk_rviz_plugins
euscollada
pr2_controllers_msgs
pr2eus
pr2eus_moveit
jsk_interactive_marker

% for pkg in $(rospack depends jsk_interactive_test); do
  rosdep resolve $pkg --rosdistro kinetic --os=ubuntu:xenial &>/dev/null || echo $pkg;
done
ml_classifiers
libccd
fcl
jsk_recognition_msgs
jsk_recognition_utils
resized_image_transport
robot_self_filter
jsk_pcl_ros_utils
rethink_ee_description
baxter_description
jsk_pcl_ros
jsk_rviz_plugins
euscollada
pr2_controllers_msgs
pr2eus
pr2eus_moveit
jsk_interactive_marker
jsk_interactive
```

longjie, 2016-11-28:
Hmm, it seems jsk_rviz_plugins have much more unmet dependencies in Kinetic, such as jsk_perception related packages. It may take some times.

Jade is OK, I could build jsk_rviz_plugins in Jade, after all jsk binary packages installed. 

Thanks anyway,

k-okada, 2016-11-28:
you do not have to compile jsk_rviz_plugins, jade is already released as
deb package
https://github.com/ros/rosdistro/pull/12852
http://repositories.ros.org/status_page/ros_jade_default.html?q=jsk_rviz_plugins&s=1

wkentaro, 2016-11-28:
@longjie For kinetic, you only need to install jsk_recognition from source.
(for jsk_rviz_plugins or jsk_rqt_plugins)
https://github.com/jsk-ros-pkg/jsk_visualization/pull/649/files#diff-a9585a05e3e573d88fd2506df53f7bee

The test on https://github.com/jsk-ros-pkg/jsk_visualization/pull/649 passes, so please see it for more detail.

longjie, 2016-11-28:
Yeah, I see. I'm closing this issue when we can build it (actually my colleague is doing). 
We may be happier sticking to Jade for a while. BTW thank you for your support!

furushchev, 2018-06-22:
`jsk_rviz_plugins` works on kinetic for now. I think it is ok to close this issue.

closed 2018-06-23 by k-okada

kargarisaac, 2019-03-05:
It has a lot of dependencies for kinetic. Every time I add the required package, it needs another one.
Is there any pipeline to do it?

knorth55, 2019-03-06:
@kargarisaac what do you mean?
if you don't want to install a lot of dependencies  again and again, it is better to make docker image for your own use case.

kargarisaac, 2019-03-06:
> @kargarisaac what do you mean?
> if you don't want to install a lot of dependencies again and again, it is better to make docker image for your own use case.

I mean when I clone one package, it needs another one, then another ... . I continued it but it didn't finish ...

knorth55, 2019-03-06:
@kargarisaac  do not use packages from source.
use debian packages.
just type `sudo apt install ros-kinetic-jsk-rviz-plugins`