jsk-ros-pkg/jsk_visualization #652: Segmentation fault when add topic in Rviz on kinetic

mizmizo, 2017-01-11:
ros kinetic環境でrvizでjsk_ros_pluginsのtopicを表示させようと（Add->display type選択->OK）すると、以下のようにsegmantation faultとなってrvizが落ちてしまいます。
```
$ rviz
[ INFO] [1484138788.341463477]: rviz version 1.12.4
[ INFO] [1484138788.341504463]: compiled against Qt version 5.5.1
[ INFO] [1484138788.341515428]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1484138789.293059399]: Stereo is NOT SUPPORTED
[ INFO] [1484138789.293152010]: OpenGl version: 4.5 (GLSL 4.5).
Segmentation fault (core dumped)
```

Rvizのデフォルトのtype(pointclud2等)では正常に表示できていて、
js_rviz_pluginsのビルドもwarning等出ずに通っているようです。

考えられる原因や確認すべき点等ご存知でしょうか？

wkentaro, 2017-01-11:
> jsk_ros_pluginsのtopicを表示させようと

というのはどういう意味ですか？
jsk_rviz_pluginsのどのdisplay typeでもsegfaultするという意味でしょうか？

mizmizo, 2017-01-11:
>jsk_rviz_pluginsのどのdisplay typeでもsegfaultするという意味でしょうか？
説明が分かりづらかったです。そのとおりで、どのtypeを選んでもsegfaultしてしまいます。

k-okada, 2017-01-11:
gdbかけてどこで落ちるか調べてみましょう

mizmizo, 2017-01-11:
```
$ gdb rviz
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.04) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from rviz...(no debugging symbols found)...done.
(gdb) run
Starting program: /opt/ros/kinetic/bin/rviz 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffdea33700 (LWP 12250)]
[New Thread 0x7fffd3f54700 (LWP 12251)]
[New Thread 0x7fffd3753700 (LWP 12252)]
[New Thread 0x7fffd2d20700 (LWP 12253)]
[New Thread 0x7fffd0f73700 (LWP 12255)]
[New Thread 0x7fffc3fff700 (LWP 12256)]
[ INFO] [1484141553.504616114]: rviz version 1.12.4
[ INFO] [1484141553.504654154]: compiled against Qt version 5.5.1
[ INFO] [1484141553.504665171]: compiled against OGRE version 1.9.0 (Ghadamon)
[New Thread 0x7fffc37fe700 (LWP 12259)]
[New Thread 0x7fffc2ffd700 (LWP 12260)]
[New Thread 0x7fffc27fc700 (LWP 12261)]
[New Thread 0x7fffc1ffb700 (LWP 12266)]
[Thread 0x7fffd0f73700 (LWP 12255) exited]
[New Thread 0x7fffd0f73700 (LWP 12267)]
[New Thread 0x7fffad36e700 (LWP 12268)]
[New Thread 0x7fffacb6d700 (LWP 12269)]
[New Thread 0x7fffa3fff700 (LWP 12270)]
[ INFO] [1484141554.445163769]: Stereo is NOT SUPPORTED
[ INFO] [1484141554.445277626]: OpenGl version: 4.5 (GLSL 4.5).
[New Thread 0x7fffa37fe700 (LWP 12285)]
[New Thread 0x7fffa2ffd700 (LWP 12286)]
[New Thread 0x7fffa1477700 (LWP 12295)]
[New Thread 0x7fffa0c76700 (LWP 12298)]
[Thread 0x7fffc3fff700 (LWP 12256) exited]

Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x00007fff7d69d5a4 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
```

ということなのでQtで落ちていて、
Kineticではqt5がサポートされているのに対してqt4.8が呼び出されているのが原因のように思われます。

バージョンを上げて解決するか試してみます。ご教授ありがとうございました。

furushchev, 2018-06-22:
It looks this issue was already solved by #662

closed 2018-06-23 by k-okada