jsk-ros-pkg/jsk_visualization #685: PolygonArray is rendered as black color.

pxlong, 2018-01-24:
Hi, thanks for sharing this great rviz plugin.

I use `jsk_recognition_msgs::PolygonArray` in Ubuntu 16.04 + ROS Kinetic. 
The polygon array are always rendered with black color in rviz when I disable "only border" and change "coloring" to Flat color, no matter what color I picked.

k-okada, 2018-01-24:
Sorry that I can not reproduce this problem
Can you try with this sample code to publish `jsk_recognition_msgs::PolygonArray` ?
```
rosrun jsk_rviz_plugins polygon_array_sample.py
```
and could visualize in rviz, I have tested in both deb (2.1.1) and latest source (2.1.3+)
[image]

pxlong, 2018-01-24:
Hi @k-okada, many thanks for the quick reply.

I recorded a video to show the problem. Please check: https://youtu.be/NufN7DZCzYU 
It will not always rendered with black, the color will change according to the viewpoint. At some viewpoints, the color seen by human's eye is black.

k-okada, 2018-01-24:
OK, though I can not see your video as follows
[image]

I think I can understand your situation, how about https://github.com/jsk-ros-pkg/jsk_visualization/pull/686 ? Will this solve your problem?
I'm not sure about your usecases, but in some situation, for example display float plane in 3D point cloud scene,current setting  will help, so I added 'enable lighting' property 

[image]
[image]

pxlong, 2018-01-25:
@k-okada, sorry for the unplayable video, I am sure you can see it now: https://youtu.be/NufN7DZCzYU.
Thanks

pxlong, 2018-01-25:
Hi @k-okada, thanks so much, #686 perfectly solve my issue.

PR #686 (2018-09-08): add enable lighitng property in polygon_array_display
Closes #685
closed 2018-01-26 by pxlong