jsk-ros-pkg/jsk_visualization #690: Convex Bounding Box

ClementLeBihan, 2018-04-12:
Hi,

I would like to know if there exist a Convex Bounding Box. I found this image on the web, and would like to use this box in RVIZ with my convex clusters. 

It's seems that it doesn't exist in the jsk_rviz_plugin, but you might be able to help me.

Thank you,
Best,
ClémentLB

[image]

ClementLeBihan, 2018-08-22:
Hi @iory !

I didn't see you answered me ! Yes I saw it was possible by using Marker but I prefer jsk because you can set LineWith, Color etc from rviz panel !

In fact I changed a little bit the way I express the polygon and now it's a nice convex-hull polygon exactly like the image !

closed 2018-08-22 by ClementLeBihan

wkentaro, 2018-08-22:
@ClementLeBihan Have you changed the rviz plugin of Marker? I think it is better to add feature to  the marker plugin to change color, line width, etc.