jsk-ros-pkg/jsk_visualization #694: find_package doesn't load jsk_rviz_plugins

ClementLeBihan, 2018-06-12:
Hi guys,

I added jsk_rviz_plugins in my find_package(catkin REQUIRED COMPONENTS),
And after that, I have all my lib except libjsk_rviz_plugins.so

And so I have a undefined reference during link ... Anyone know how to link to jsk_rviz_plugin ? I need to create an overlayer of jsk_rviz_plugins :)

Best,
Clément

k-okada, 2018-06-13:
can you share your package, so that I can help your debugging?

iory, 2018-07-23:
@ClementLeBihan 

Please share your package, or test case. I'll debug it.

ClementLeBihan, 2018-08-21:
find_package(catkin REQUIRED COMPONENTS
  rviz
  jsk_recognition_msgs
  jsk_topic_tools
  jsk_rviz_plugins
  cv_bridge)

# Use ccache if installed to make it fast to generate object files
find_program(CCACHE_FOUND ccache)
if(CCACHE_FOUND)
  set_property(GLOBAL PROPERTY RULE_LAUNCH_COMPILE ccache)
  set_property(GLOBAL PROPERTY RULE_LAUNCH_LINK ccache)
endif(CCACHE_FOUND)

catkin_package(
    DEPENDS rviz
    CATKIN_DEPENDS rviz
    jsk_recognition_utils cv_bridge jsk_topic_tools jsk_rviz_plugins
    INCLUDE_DIRS 
    LIBRARIES 
)

message(" catkin_LIBRARIES : " ${catkin_LIBRARIES})

ClementLeBihan, 2018-08-21:
<?xml version="1.0"?>
<package>
  <name>TEST</name>
  <version>2.1.3</version>
  <license>TODO</license>

  <buildtool_depend>catkin</buildtool_depend>

  <build_depend>cv_bridge</build_depend>
  <build_depend>jsk_rviz_plugins</build_depend>
  <build_depend>jsk_recognition_msgs</build_depend>
  <build_depend>jsk_recognition_utils</build_depend>
  <build_depend>jsk_topic_tools</build_depend>
  <build_depend>rosbuild</build_depend>
  <build_depend>rviz</build_depend>

  <run_depend>cv_bridge</run_depend>
  <run_depend>jsk_rviz_plugins</run_depend>
  <run_depend>jsk_topic_tools</run_depend>
  <run_depend>jsk_recognition_msgs</run_depend>
  <run_depend>jsk_recognition_utils</run_depend>
  <run_depend>rviz</run_depend>

  <export>
    <rviz plugin="${prefix}/plugin_description.xml"/>
  </export>

</package>

iory, 2018-08-21:
Hmm, Can you share a simple package of your project?

ClementLeBihan, 2018-08-21:
Unfortunately, it's professionnal so I can't share any code.

But I found how to solve the lib issue : In the jsk_rviz_pluginsConfig.cmake, I have to change 
set(libraries "") to set(libraries "jsk_rviz_plugins") !! line 119

Best,
Clément

iory, 2018-08-21:
OK.

>  In the jsk_rviz_pluginsConfig.cmake, I have to change

This is not good. 
In your CMakeLists.txt's target_link_libraries, you should add ```${catkin_LIBRARIES}```. 
You should refer ```jsk_interactive_marker``` package because this package is using jsk_rviz_plugins and link it.
https://github.com/jsk-ros-pkg/jsk_visualization/blob/master/jsk_interactive_markers/jsk_interactive_marker/CMakeLists.txt

iory, 2018-08-21:
Also add following code to your CMakeLists.txt

```
include_directories(include ${catkin_INCLUDE_DIRS})
link_directories(${catkin_LIBRARY_DIRS})
```

ClementLeBihan, 2018-08-21:
It doesn't work ... I upload a test : https://github.com/ClementLeBihan/rviz_layer

Tell me if you can compile !

Best,
Clément

iory, 2018-08-21:
Please try following steps.

```
mkdir -p ~/catkin_ws/rviz_layer/src
cd ~/catkin_ws/rviz_layer/src
git clone https://github.com/jsk-ros-pkg/jsk_visualization
git clone https://github.com/ClementLeBihan/rviz_layer
cd ~/catkin_ws/rviz_layer
rosdep install --from-paths -y -r src
catkin build
```

ClementLeBihan, 2018-08-21:
It works, that's a solution to compile, but it doesn't solve the problem that when you use the ros-kinetic-jsk-rviz-plugins package and you put  jsk_rviz_plugins in your catkin dependencies in your CMakelist the lib isn't linked...

Do you know why the line 119 of the jsk_rviz_pluginsConfig.cmake is set(libraries "") instead of set(libraries "jsk_rviz_plugins") ?

Best,
Clément

iory, 2018-08-21:
From https://github.com/jsk-ros-pkg/jsk_visualization/pull/696 , jsk_rviz_plugins add library to catkin_package.
So you can check the jsk_rviz_pluginsConfig.cmake and you can see set(libraries "jsk_rviz_plugins") in .cmake file.

ClementLeBihan, 2018-08-21:
Thank you so much @iory, that's exacly the answer I was looking for :) I'm sorry I didn't see this issue.

BTW, do you know how to add manually (with the path)  libjsk_rviz_plugins.so to the catkin_LIBRARIES in my CMakeList ?

Best,
Clément

iory, 2018-08-21:
I send PR to your repository.
https://github.com/ClementLeBihan/rviz_layer/pull/1
You can compile it without source of jsk_rviz_plugins.

iory, 2018-08-22:
Your welcome.
We also very interested in your rviz_plugin based on our package and waiting for your PR!

BTW, I am glad if you reply to this issue  https://github.com/jsk-ros-pkg/jsk_visualization/issues/690

kargarisaac, 2019-03-06:
> rosdep install --from-paths -y -r src

Hi,
When I run ``` rosdep install --from-paths -y -r src``` I get:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
jsk_rviz_plugins: No definition of [view_controller_msgs] for OS [debian]
rviz_layer: No definition of [rviz] for OS [debian]
jsk_visualization: No definition of [jsk_interactive_test] for OS [debian]
jsk_rqt_plugins: No definition of [rostest] for OS [debian]
jsk_interactive_marker: No definition of [rviz] for OS [debian]
jsk_interactive_test: No definition of [jsk_interactive_marker] for OS [debian]
jsk_interactive: No definition of [jsk_interactive_marker] for OS [debian]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```
I spent a looooooooot of time to install jsk_rviz_plugins for kinetic but I couldn't. How can I do that?

knorth55, 2019-03-06:
@kargarisaac  `rosdep update` `apt update` `apt install ros-jsk-rviz-plugins`.

kargarisaac, 2019-03-06:
> apt install ros-jsk-rviz-plugins

Thank you. But When I run ```sudo apt install ros-jsk-rviz-plugins``` I get:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-jsk-rviz-plugins
```

I have this problem for some other jsk packages too: unable to locate ...

knorth55, 2019-03-06:
it is typo. type `sudo apt install ros-kinetic-jsk-rviz-plugins`

kargarisaac, 2019-03-06:
> sudo apt install ros-kinetic-jsk-rviz-plugins

same reult:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-jsk-rviz-plugins
```

kargarisaac, 2019-03-06:
I installed it many times ago. But I don't remember it. Now that I want to install it I cannot. I want to use [this](https://github.com/CPFL/Autoware/tree/master/ros/src/computing/perception/detection/lidar_detector/packages/lidar_euclidean_cluster_detect) package from Autoware on my system.

knorth55, 2019-03-06:
have you installed `ROS`?
please read this wiki. http://wiki.ros.org/kinetic/Installation/Ubuntu
it seems that your apt does not have current `ROS` package release information.
please try to reconfigure apt package for ROS again.

```
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116
sudo apt update
sudo apt install ros-kinetic-jsk-rviz-plugins
```

knorth55, 2019-03-06:
btw, which OS and OS version are you using?

kargarisaac, 2019-03-06:
@knorth55 
Thanks a lot. It works. I use ubuntu 16.04.

ruman23, 2024-04-23:
For me the follwoing command worked. 

sudo apt-get install ros-$ROS_DISTRO-jsk-rviz-plugins