jsk-ros-pkg/jsk_visualization #697: Polygons are not solid

chinitaberrio, 2018-06-21:
Hi all!

I'm publishing 121 polygons in a polygon array, in rviz i can only see the border of the polygon (here the only border box is ticked)

[image]

When the "only border" box is not ticked i can't see anything regarding the polygons.

[image]

Any idea of what could be the issue?

S.

furushchev, 2018-06-21:
@chinitaberrio Could you paste a content of the published message for further debugging? (If this problem happens even the number of polygons is only one, then please paste the message on that case)

chinitaberrio, 2018-06-21:
Here the published message.
[output.txt](https://github.com/jsk-ros-pkg/jsk_visualization/files/2122525/output.txt)

Thanks.

furushchev, 2018-06-21:
@chinitaberrio Could you also set `frame_id` of a header for each polygons members and check if it works?

chinitaberrio, 2018-06-21:
It worked! thanks!

furushchev, 2018-06-21:
@chinitaberrio 
Great!
I'll create a pull request to show error on this situation later to let users notice easier.

PR #698 (2018-06-23): jsk_rviz_plugins: warn on missing frame_id
Closes #697 
In this PR, a warning is displayed when received messages are not valid.

[image]
closed 2018-06-23 by k-okada