jsk-ros-pkg/jsk_visualization #702: Melodic/bionic support

jorritsmit, 2018-08-07:
I was wondering if melodic/bionic integration is on the horizon?

k-okada, 2018-08-08:
ok, we're start testing on this at https://github.com/jsk-ros-pkg/jsk_visualization/pull/703
currently we need release of following package before jsk_visualization.
```
+ rosdep install -q -y --rosdistro melodic -n -q -r --ignore-src --from-paths /opt/ros/melodic/share .

ERROR: the following packages/stacks could not have their rosdep keys resolved

to system dependencies:

jsk_interactive_marker: Cannot locate rosdep definition for [jsk_recognition_utils]

jsk_interactive: Cannot locate rosdep definition for [dynamic_tf_publisher]

jsk_rqt_plugins: No definition of [python-sklearn] for OS version [bionic]

jsk_rviz_plugins: Cannot locate rosdep definition for [view_controller_msgs]

Continuing to install resolvable dependencies...
```

jorritsmit, 2018-08-08:
Ok good to hear. I tried this as well and build the packages you mentioned from source. Building jsk_visualization gives opencv errors. Good luck and let me know if you need help!

k-okada, 2018-08-08:
can you share your error messages?

jorritsmit, 2018-08-08:
actually the only problem is the splitting of opencv highgui into 2 libraries. I made a pull request on jsk_common_msgs, but it has to be done with a lot of the jsk plugins. i don't have time to make the pull requests now, but i can do it tonight if you want?

jorritsmit, 2018-08-08:
btw see https://docs.opencv.org/3.1.0/db/dfa/tutorial_transition_guide.html

k-okada, 2018-08-09:
> guide.html

sure , pull requests are always welcome

jorritsmit, 2018-08-10:
as mentioned in https://github.com/jsk-ros-pkg/jsk_common_msgs/pull/22, there was something so jsk packages succeed on bionic. I'll close the issues since #703 covers this

closed 2018-08-10 by jorritsmit