jsk-ros-pkg/jsk_visualization #711: [jsk_interactive_marker] Melodic build failure

knorth55, 2018-09-28:
Related to #710 

workspace rosinstall

```
- git:
    local-name: jsk_visualization
    uri: https://github.com/jsk-ros-pkg/jsk_visualization.git
- tar:
    local-name: jsk_recognition/jsk_recognition_utils
    uri: https://github.com/tork-a/jsk_recognition-release/archive/release/kinetic/jsk_recognition_utils/1.2.5-0.tar.gz
    version: jsk_recognition-release-release-kinetic-jsk_recognition_utils-1.2.5-0
- tar:
    local-name: jsk_recognition/jsk_recognition_msgs
    uri: https://github.com/tork-a/jsk_recognition-release/archive/release/kinetic/jsk_recognition_msgs/1.2.5-0.tar.gz
    version: jsk_recognition-release-release-kinetic-jsk_recognition_msgs-1.2.5-0
- tar:
    local-name: jsk_common/jsk_topic_tools
    uri: https://github.com/tork-a/jsk_common-release/archive/release/kinetic/jsk_topic_tools/2.2.7-0.tar.gz
    version: jsk_common-release-release-kinetic-jsk_topic_tools-2.2.7-0
- tar:
    local-name: jsk_common/dynamic_tf_publisher
    uri: https://github.com/tork-a/jsk_common-release/archive/release/kinetic/dynamic_tf_publisher/2.2.7-0.tar.gz
    version: jsk_common-release-release-kinetic-dynamic_tf_publisher-2.2.7-0
```

knorth55, 2018-09-28:
error log is here. https://drive.google.com/file/d/19lzzsMCRwU9Z3TFPNmsLfY18Ym2jy2om/view?usp=sharing

knorth55, 2018-10-26:
This is mainly caused change in urdfdom 1.0.0, which is default in Melodic.
@k-okada Do you have any ideas?
https://github.com/ros/urdfdom_headers/blob/1.0.0/urdf_model/include/urdf_model/types.h#L44-L46

k-okada, 2018-10-28:
see https://github.com/PR2/pr2_common/pull/272/commits/e1196348068e3bdcb0935142d9bda18a1f40fa8b

knorth55, 2018-11-06:
Closed by #710

closed 2018-11-06 by knorth55