jsk-ros-pkg/jsk_visualization #712: ビルドエラー

KazukiHiraizumi, 2018-10-01:
catkin_makeが以下のエラーとなります。
~~~
-- Could not find the required component 'jsk_hark_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "jsk_hark_msgs"
  with any of the following names:

    jsk_hark_msgsConfig.cmake
~~~
RosはKineticです。

k-okada, 2018-10-01:
rワークスペースで
rosdep install --from-paths src --ignore-src
として必要なパッケージをインストールしてください．

ただ，基本的にはコンパイルせずとも
sudo apt-get install ros-kinetic-jsk-rviz-config
で使えるとおもいます．

KazukiHiraizumi, 2018-10-01:
OverlayTextを使わせて頂きたかったので、以下でインストールできました
~~~
sudo apt-get install ros-kinetic-jsk-visualization
sudo apt-get install ros-kinetic-jsk-perception
~~~
OverlayTextのExampleは以下で起動できました
~~~
roslaunch jsk_rviz_plugins overlay_sample.launch
~~~
ありがとうございました

iory, 2018-10-05:
@KazukiHiraizumi 
問題がなければこちらのissueをcloseお願いします。

closed 2018-10-05 by KazukiHiraizumi