jsk-ros-pkg/jsk_visualization #713: Crash when using CameraInfo display

YoanSallami, 2018-10-04:
hello,

when using the camera info display, I have this message :
`̀``
rviz: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/include/OgreAxisAlignedBox.h :252 : void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&):  l'assertion « (min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner" » a échoué.
Abandon (core dumped)
```
Even when using the static_virtual_camera.py

Any idea on how to manage that ?

iory, 2018-10-05:
Thank you for your report.

How to install jsk_rviz_plugins? By installing ```sudo apt install ros-kinetic-jsk-rviz-plugins```?

k-okada, 2018-10-05:
if you're using
`/opt/ros/kinetic/lib/jsk_recognition_utils/static_virtual_camera.py`, you
need valid camera info parameters,
```
header:
  seq: 272
  stamp:
    secs: 1538704752
    nsecs: 969589948
  frame_id: "static_virtual_camera"
height: 768
width: 1024
distortion_model: ''
D: []
K: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
R: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
P: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
```
--
◉ Kei Okada

iory, 2018-10-05:
rviz plugin of ```CameraInfo``` should subscribe valid camera_info.
But crash is not for users. So I'll create PR to this error.

Please download following bag file and rviz file and try following commands.
https://drive.google.com/file/d/1g-6bfkOrJ3pPR8Ge-cD6gKkLYPD7pUav/view?usp=sharing
https://drive.google.com/file/d/1VNMtp9sEnqP0YDyK2JcCRImD48VhiJ_o/view?usp=sharing

```
roscore
rosbag play 2018-10-05-17-50-38.bag --loop --clock
rviz -d camera_info.rviz
```

YoanSallami, 2018-10-05:
It worked with the ros bag... Si it is not possible to get it worked for uncalibrated cameras ?

iory, 2018-10-06:
We can calculate intrinsic parameters from fov.
Please try following code and run.

https://github.com/iory/jsk_recognition/blob/7eb85b2b65de749cd96d6a67ca9e24e305a45051/jsk_perception/node_scripts/image_publisher.py

```
python image_publisher.py _file_name:=/opt/ros/kinetic/share/jsk_perception/sample/kiva_pod_image_color.jpg _encoding:=bgr8 _publish_info:=true _fovx:=83.1 _fovy:=53.8
```

[image]

iory, 2018-10-06:
@YoanSallami 

You can publish image with valid camera info with following PR.
https://github.com/jsk-ros-pkg/jsk_recognition/pull/2340