jsk-ros-pkg/jsk_visualization #719: Target visualize failed on kinetic.

tdnet12434, 2018-10-31:
Hi, thank you for great visualize tool.
This help me a lot to visualize my swarm robot.
But today i've tried to use target visualize but some time have failed to add it on rviz.
The error is show below

`
rviz: /usr/include/boost/smart_ptr/shared_ptr.hpp:648: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = jsk_rviz_plugins::FacingObject; typename boost::detail::sp_member_access<T>::type = jsk_rviz_plugins::FacingObject*]: Assertion `px != 0' failed.
`
My machine is ubuntu 16.04 running kinetic.
Thank you advanced

iory, 2018-11-01:
Thank you for using jsk_visualization and your issue report!

Please also teach me version of ```jsk_rviz_plugins```.

for example
```
sudo dpkg --list  | grep jsk-rviz-plugins
```

tdnet12434, 2018-11-01:
The version of `jsk_rviz_plugins` is 

`ii  ros-kinetic-jsk-rviz-plugins                                2.1.3-0xenial-20180805-214048-0800                          amd64        The jsk_rviz_plugins package`

Thank you advanced krub.

iory, 2018-11-01:
OK. 
I create the PR to fix this error.
https://github.com/jsk-ros-pkg/jsk_visualization/pull/720

Could you check the PR by following commands.
```
mkdir -p ~/catkin_ws/jsk_rviz_plugins/src
git clone -b target-visualize https://github.com/iory/jsk_visualization ~/catkin_ws/jsk_rviz_plugins/src/jsk_visualization
cd ~/catkin_ws/jsk_rviz_plugins
rosdep install --from-paths --ignore-src -y -r src
catkin b jsk_rviz_plugins
source ~/catkin_ws/jsk_rviz_plugins/devel/setup.bash
rviz 
```

tdnet12434, 2018-11-01:
Thank you! @iory it's worked! 
I can close this issue now krub.

closed 2018-11-01 by tdnet12434