jsk-ros-pkg/jsk_visualization #722: Change size of pictogram

norala22, 2018-11-28:
I am currently trying to use pictograms from FontAwesome in my Rviz visualization.  
Is it possible to change the size of the icon/pictogram? I tried all possible fields (including .size) to change the size of the pictogram, but nothing worked. Thanks

iory, 2019-01-30:
Thanks for your issue report!
Could you let me know your environment?

In my environment 
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.5 LTS
Release:        16.04
Codename:       xenial
$ sudo dpkg --list | grep jsk-rviz-plugins
ii  ros-kinetic-jsk-rviz-plugins                                2.1.4-0xenial-20181117-040432-0800                    amd64        The jsk_rviz_plugins package
```
I can successfully changed size of pictograms.

### size == 1

[image]

### size == 2

[image]

mqcmd196, 2023-05-07:
Close there has been no movement for a while.

closed 2023-05-07 by mqcmd196