jsk-ros-pkg/jsk_visualization #724: ros-melodic-jsk-rviz availability?

phil333, 2018-12-17:
Hi, I m using your jsk-rviz tools and i was wondering if you are planning to make them available for melodic via sudo apt-get?

k-okada, 2018-12-17:
Sure, we’re working on this 
https://github.com/jsk-ros-pkg/jsk_recognition/issues/2358#issuecomment-436465484

christian-rauch, 2019-02-11:
@k-okada Any news on this? `jsk_common` has been released to melodic.
This is also blocking `jsk_visualization` from being released to melodic: http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/bionic-travis-jsk-ros-pkg-jsk-recognition-melodic-deb-true-true/19/console

k-okada, 2019-02-20:
This has been released -> https://github.com/ros/rosdistro/pull/20334

iory, 2019-03-27:
Close because this issue seems to be solved.

closed 2019-03-27 by iory