jsk-ros-pkg/jsk_visualization #734: Text on 3D bounding box

jinfagang, 2019-04-15:
Redirect from this issue: https://github.com/jsk-ros-pkg/jsk_recognition/issues/2407

I want draw some text on bounding box like this:
[image]

Is this avaiable in jsk? Someone kindly tell me how to add it? I can not found such a field to draw text on bounding box

708yamaguchi, 2019-04-15:
I have not tested yet, but you can use `jsk_rviz_plugins/Pictogram` to show the string popups.

The document is here:
https://jsk-visualization.readthedocs.io/en/latest/jsk_rviz_plugins/plugins/pictogram.html

jinfagang, 2019-04-15:
@708yamaguchi Why not just add a field in BoundingBoxAarray?

708yamaguchi, 2019-04-15:
I think that separating `BoundingBoxArray` and `Pictogram` is better.

This is because it is very time-consuming to add the pictogram field  to many msgs for visualization. (e.g. there are `BoundingBox`, `Marker`, `Polygon`, `Pose` and so on)

knorth55, 2019-04-15:
@jinfagang if you have suggestion, please make PR. let's discuss then.