jsk-ros-pkg/jsk_visualization #738: RViz polygon blinking

stribor14, 2019-05-09:
Hello, I'm using the package version from Ubuntu 16.04 PPA (version 2.1.5-0xenial-20190320-184807-0800)

I'm trying to visualize the area which laser scan is covering with a polygon, and I'm using jsk polygon array with single polygon because your rviz plugin allows polygon to be filled with color.

When "only border" is selected, visualization behaves normally:

[image]

 But when a polygon is shown filled, everything starts to blink while randomly connecting some parts of a polygon (you can see overlap with transparency turned on):

[image]

knorth55, 2019-05-09:
can you give us rosbag file to reproduce this issue or can you tell us how to reproduce your issue?

stribor14, 2019-05-13:
Sorry, I changed the code, but the node was really simple. I've taken readings from the laser_scan topic of nav350 simulated in the Gazebo, then I've created a polygon in which I inserted all of the points laser has seen (range * sin/cos (angle)), sorted by angle. Polygons header is the same as a laser_scan header because I wanted 
same frame_id and timestamp.

knorth55, 2019-05-13:
I have no idea because we cannot reproduce your problem in our environment.
but it seems that your sorting might be incorrect.

stribor14, 2019-08-27:
This is what probably also happened to me, but it looked like blinking because polygon would change slightly change every iteration, so those mistakes jumped around

YutoUchimi, 2019-08-27:
Although polygon in rviz does not blink, it is filled wrongly in some cases.

### case 1
```
<launch>
  <node name="polygon_array_publisher"
        pkg="jsk_recognition_utils" type="polygon_array_publisher.py">
    <rosparam>
      publish_rate: 10.0
      frame_id: map
      polygons:
        - points:
            - [3, 0, 0]
            - [4, 1, 0]
            - [1, 1, 0]
            - [1, 4, 0]
            - [0, 3, 0]
            - [-1, 4, 0]
            - [-1, 1, 0]
            - [-4, 1, 0]
            - [-3, 0, 0]
            - [-4, -1, 0]
            - [-1, -1, 0]
            - [-1, -4, 0]
            - [0, -3, 0]
            - [1, -4, 0]
            - [1, -1, 0]
            - [4, -1, 0]
          label: 0
          likelihood: 1.0
    </rosparam>
  </node>
</launch>
```
[image]
[image]

### case 2
```
<launch>
  <node name="polygon_array_publisher"
        pkg="jsk_recognition_utils" type="polygon_array_publisher.py">
    <rosparam>
      publish_rate: 10.0
      frame_id: map
      polygons:
        - points:
            - [3, 0, 0]
            - [4, 1, 0]
            - [1, 1, 0]
            - [1, 4, 0]
            - [0, 3, 0]
            - [-1, 4, 0]
            - [-1, 1, 0]
            - [-4, 1, 0]
            - [-3, 0, 0]
          label: 0
          likelihood: 1.0
    </rosparam>
  </node>
</launch>
```
[image]
[image]

closed 2020-02-12 by k-okada