jsk-ros-pkg/jsk_visualization #742: Segmentation fault when add topic in Rviz on kinetic

Kai1991, 2019-06-23:
`
gdb rviz
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from rviz...(no debugging symbols found)...done.
(gdb) run
Starting program: /opt/ros/kinetic/bin/rviz 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffdd533700 (LWP 1739)]
[ INFO] [1561276852.504446238]: rviz version 1.12.15
[ INFO] [1561276852.504529741]: compiled against Qt version 5.5.1
[ INFO] [1561276852.504539870]: compiled against OGRE version 1.9.0 (Ghadamon)
[New Thread 0x7fffd7fff700 (LWP 1742)]
[New Thread 0x7fffd77fe700 (LWP 1743)]
[New Thread 0x7fffd6ffd700 (LWP 1744)]
[New Thread 0x7fffd67fc700 (LWP 1749)]
[New Thread 0x7fffc4607700 (LWP 1751)]
[New Thread 0x7fffc3e06700 (LWP 1752)]
[New Thread 0x7fffc3605700 (LWP 1753)]
[New Thread 0x7fffc2b04700 (LWP 1754)]
[New Thread 0x7fffc2303700 (LWP 1755)]
[New Thread 0x7fffc1b02700 (LWP 1756)]
[New Thread 0x7fffbfbde700 (LWP 1757)]
[New Thread 0x7fffbf3dd700 (LWP 1758)]
[New Thread 0x7fffbebdc700 (LWP 1759)]
[ INFO] [1561276853.240531925]: Stereo is NOT SUPPORTED
[ INFO] [1561276853.240681850]: OpenGl version: 3 (GLSL 1.3).
[New Thread 0x7fffbe3db700 (LWP 1772)]
[New Thread 0x7fffbdbda700 (LWP 1773)]
[New Thread 0x7fffa6a2a700 (LWP 1789)]
[New Thread 0x7fffa6229700 (LWP 1790)]

Thread 1 "rviz" received signal SIGSEGV, Segmentation fault.
0x00007ffff43c1dda in ros::TimeBase<ros::Time, ros::Duration>::operator+(ros::Duration const&) const () from /opt/ros/kinetic/lib/librostime.so
`

knorth55, 2019-06-23:
It seems your rviz is crashing because of `librostime.so`.
maybe it is related to this issue https://github.com/ros/geometry/issues/176#issuecomment-423693686
can you try `sudo apt-get update; sudo apt-get upgrade` and run `rviz` again?

Kai1991, 2019-06-24:
> It seems your rviz is crashing because of `librostime.so`.
> maybe it is related to this issue [ros/geometry#176 (comment)](https://github.com/ros/geometry/issues/176#issuecomment-423693686)
> can you try `sudo apt-get update; sudo apt-get upgrade` and run `rviz` again?

thanks .

closed 2019-06-24 by Kai1991