jsk-ros-pkg/jsk_visualization #745: [jsk_rviz_plugins/video_capture] Fix frame rate

darkdragon-001, 2019-07-22:
Some frames seem to be dropped and thus the video plays much faster than how reality happened.

knorth55, 2019-07-22:
is this feature request?
are you planning to add this feature?

darkdragon-001, 2019-07-23:
@knorth55 It's a bug report. Unfortunately, I don't have time at the moment to look into it.

YutoUchimi, 2019-09-03:
In current implementation, a captured image is written to output video at every rendering loop on rviz.
So unless rviz rendering fps luckily matches video fps you set, you will see fast/slow play on recorded video, though output video fps is accurately set by this plugin.

A possible solution is to create ros::WallTimer and write images to video in the callback.
It may improve the frame rate a little bit, but accurate frame rate is difficult to be achieved because it's not real-time.

How about using an external screencast tool like kazam?

darkdragon-001, 2019-09-03:
@YutoUchimi Thanks for your insights on the current implementation. 

In the end, I used a screencast tool with a lot of manual work. I had a folder of about 200 rosbags which I wanted to compare. They need the exact same resolution, perspective and frame rate. Doing this manually was white time-consuming. 
Since I use rosbags, I don't need *real-time*, but the *same* speed for all of them.

For normal graphs, I implemented my own rqr_plot like tool, which advanced time in the rosbag for the amount of time one frame needs. If it's possible to manually feed rviz with ROS messages and then call some screenshot function, this would perfectly solve my problem.