jsk-ros-pkg/jsk_visualization #772: Noetic release

reinzor, 2020-05-25:
Will this package be released for Noetic anytime soon :)?

knorth55, 2020-05-26:
we are working on it, but there are several packages which `jsk_visualization` depends on but not released in Noetic.
It does not seem that the release on Noetic is soon (I mean not this week).
If you really want to use this now on Noetic, you need to solve dependency by yourself (it would be really troublesome).

ana-GT, 2020-06-21:
Hi, our team also uses jsk_visualization (specifically jsk_rviz_plugins), and we are very much looking forward to see it released in noetic. I recently upgraded to 20.04 and am in the process of testing our tools. Is there any update on the release of this package in noetic anytime soon? (writing after reading the message above mentioning you guys were already working on it some weeks ago).

Thank you very much for your work and looking forward to use jsk_rviz_plugins soon in noetic! :)

PeterMitrano, 2020-09-28:
From what I can see, the only missing dependencies are 
 - [ ] jsk_recognition
 - [ ] view_controller_msgs

knorth55, 2020-10-27:
https://github.com/ros/rosdistro/pull/27143
Noetic release is merged.

closed 2020-10-28 by knorth55