jsk-ros-pkg/jsk_visualization #779: Will jsk_visualization be ported to ROS2?

paulerikf, 2020-09-08:
Hey there, hope this is an o.k. place to ask this... any word on whether there will be a version of jsk visualization that will work with rviz2 / ros2?

Hard to imagine life without it <3

k-okada, 2020-09-09:
Currently we have no plan/resource to port to ROS2, but volunteers are
welcome
--
◉ Kei Okada

fuhuan26, 2020-10-13:
Yeah, it will be great to have rviz2 compatible jsk_visualizaiton. The tool is so handy!

arjunskumar, 2021-03-17:
any updates on this?

ottojo, 2022-08-30:
Hi! We ported the text-overlay part to ROS2, and published the package here: https://github.com/teamspatzenhirn/overlay_rviz_plugins We intend to also port some of the other "overlay" plugins. The actually required code changes are rather minimal, and should all be contained in this commit, this might be helpful if someone decides to port the entire package: https://github.com/teamspatzenhirn/overlay_rviz_plugins/commit/5d52b380b67d0487ab321f34a9d31f261e47e685 (which we are not interested in currently)

mqcmd196, 2026-09-05:
close via https://github.com/jsk-ros-pkg/jsk_visualization/pull/903

closed 2026-09-05 by mqcmd196