jsk-ros-pkg/jsk_visualization #781: Can "jsk_rviz_plugin/PieChart' change rotation direction?

fuhuan26, 2020-09-22:
Hi, 
The default rotation direction for pie chart is counterclockwise. I am wondering if there is a way to change it to clockwise.
As I would like to use it to mimic vehicle speedometer, which is more intuitive in clockwise.
[image]

knorth55, 2020-09-23:
this is because we develop this plugin for decreasing values such as battery capacity or voltage.
but I think it is good to add new option like `rotate direction` for this rviz plugin.
We welcome for your contribution, and we look forward to seeing a PR for the new feature.

fuhuan26, 2020-09-23:
> this is because we develop this plugin for decreasing values such as battery capacity or voltage.
> but I think it is good to add new option like `rotate direction` for this rviz plugin.
> We welcome for your contribution, and we look forward to seeing a PR for the new feature.

Thank you for your reply, @knorth55. Could you please point me which file is for the current piechart implementation? I am happy to take a look and work on adding the direction selection feature. Thanks!

knorth55, 2020-09-23:
@fuhuan26 how about this option #782 ?

fuhuan26, 2020-09-24:
> @fuhuan26 how about this option #782 ?

@knorth55 That looks great!  Really appreciate your prompt help! One last question, I installed the package using `sudo apt install ros-melodic-jsk-rviz-plugins`  And I was not able to locate the pie_chart_display.cpp file and apply the change. 

Can you please show me where to find that file and then how to recompile the package? :)

Thanks a lot!

pazeshun, 2020-09-24:
@fuhuan26 You can install `jsk_rviz_plugins` from source:
```bash
cd /path/to/your/catkin_ws/src  # In the following, I use "catkin build" command. If you have a workspace using "catkin build", you can use that workspace. If you have a workspace using "catkin_make" and don't mind building time, you can use that workspace on condition that you use "catkin_make" instead of "catkin build". If you don't have a workspace, please create it by "mkdir -p /path/to/your/catkin_ws/src" and I recommend using "catkin build" in the following.
git clone https://github.com/knorth55/jsk_visualization.git
cd jsk_visualization
git checkout pie-chart-clockwise
cd /path/to/your/catkin_ws
rosdep update
rosdep install --from-paths src --ignore-src
sudo apt install python-catkin-tools  # If you use "catkin build" for the first time
catkin build jsk_rviz_plugins  # or just "catkin_make" if you choose to use "catkin_make"
source /path/to/your/catkin_ws/devel/setup.bash  # This is required on every terminal launching rviz
```
FYI
https://answers.ros.org/question/252478/how-to-build-a-package-from-source-on-ubuntu-mate-1604-lts-and-ros-kinetic/
https://catkin-tools.readthedocs.io/en/latest/verbs/catkin_build.html

fuhuan26, 2020-09-24:
> @fuhuan26 You can install `jsk_rviz_plugins` from source:
> 
> ```shell
> cd /path/to/your/catkin_ws/src  # In the following, I use "catkin build" command. If you have a workspace using "catkin build", you can use that workspace. If you have a workspace using "catkin_make" and don't mind building time, you can use that workspace on condition that you use "catkin_make" instead of "catkin build". If you don't have a workspace, please create it by "mkdir -p /path/to/your/catkin_ws/src" and I recommend using "catkin build" in the following.
> git clone https://github.com/knorth55/jsk_visualization.git
> cd jsk_visualization
> git checkout pie-chart-clockwise
> cd /path/to/your/catkin_ws
> rosdep update
> rosdep install --from-paths src --ignore-src
> sudo apt install python-catkin-tools  # If you use "catkin build" for the first time
> catkin build jsk_rviz_plugins  # or just "catkin_make" if you choose to use "catkin_make"
> source /path/to/your/catkin_ws/devel/setup.bash  # This is required on every terminal launching rviz
> ```
> 
> FYI
> https://answers.ros.org/question/252478/how-to-build-a-package-from-source-on-ubuntu-mate-1604-lts-and-ros-kinetic/
> https://catkin-tools.readthedocs.io/en/latest/verbs/catkin_build.html

@pazeshun That solved my problem. The new pie chart looks great! Thank you so much for your help!  どうもありがとうございました!

closed 2020-09-24 by fuhuan26