jsk-ros-pkg/jsk_visualization #787: jsk_visualization readthedoc is not updated since 2018.12.18

knorth55, 2020-11-03:
jsk_visualization readthedoc is not updated sincd 2018.12.18
https://jsk-visualization.readthedocs.io/en/latest/
https://readthedocs.org/projects/jsk-visualization/builds/

k-okada, 2020-11-03:
@garaemon is the owner of https://readthedocs.org/projects/jsk-visualization/ and we could not configure that. @garaemon, Could you add me as maintainer? or remove the project?

knorth55, 2020-11-17:
@garaemon hi, I also want to give us permission for jsk-visualization readthedocs project.

garaemon, 2020-11-24:
Sorry for my late response.

I've just removed the jsk-visualization project from readthedocs.
Could you create a new version of jsk-visualization on readthedocs?

k-okada, 2020-11-25:
@garaemon thank you!! created enw one https://jsk-visualization.readthedocs.io/en/latest/index.html

closed 2020-11-25 by knorth55