jsk-ros-pkg/jsk_visualization #799: RViz crashes when trying to set the "only border" parameter to false for the message jsk_rviz_plugin/PolygonArray

dymons, 2021-01-12:
Hi. The RViz config was configured for rendering using the _jsk_rviz_plugin/PolygonArray_ message on Ubuntu 18.04 ROS Melodic. When switching to Ubuntu 20.04 ROS Noetic RViz started to crash with the message:
```
[ WARN] [1610455053.000758141, 1941.347000000]: OGRE EXCEPTION(5:ItemIdentityException): Cannot find a group named rviz in ResourceGroupManager::isResourceGroupInitialised at /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 1916)
terminate called after throwing an instance of 'Ogre::ItemIdentityException'
  what():  OGRE EXCEPTION(5:ItemIdentityException): Cannot find a group named rviz in ResourceGroupManager::isResourceGroupInitialised at /build/ogre-1.9-kiU5_5/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreResourceGroupManager.cpp (line 1916)
```
Experimentally, I found that this happens when I try to set the **only border** parameter to **false** for the _jsk_rviz_plugin/PolygonArray_ message. Could you tell me how to solve this problem? Since I need to render the filled polygons, not their borders.

KenYN, 2021-07-21:
What's the latest on this issue? It's a real showstopper for me. "Only border" setting doesn't seem to make any difference, though.

knorth55, 2021-07-21:
can you investigate more detailed information?
we do not have enought resource to investigate this issue on noetic.

knorth55, 2021-08-05:
I submit a PR #820 to solve this issue.
can anyone check this PR?

closed 2021-09-15 by k-okada