jsk-ros-pkg/jsk_visualization #801: [jsk_rviz_plugins/overlay_text_display.cpp] Don't color the text's shadow

xqrkushnir, 2021-01-22:
The colour of the shadow gets affected by HTML tags, resulting in a blurry-appealing text. 

As example, passing an `OverlayText()` message with `text` field being `'<span style="color: gray;">TEST_STRING</span>'` will colour a specific part of the overlay in gray, but would also paint the shadow. See image bellow: 
[image]

The part where the background shadow is being generated:
https://github.com/jsk-ros-pkg/jsk_visualization/blob/master/jsk_rviz_plugins/src/overlay_text_display.cpp#L270-L272

xqrkushnir, 2021-01-22:
I suggest making this regex pattern check: `color.+?;`. 
Would strip out only `color: #FFFFFF;` out of something like `<span style="color: #FFFFFF;">test</span>`

knorth55, 2021-01-22:
can you check this new feature?
https://github.com/jsk-ros-pkg/jsk_visualization/pull/775

xqrkushnir, 2021-01-22:
> can you check this new feature?
> #775

Sure, I know it. But I don't intend on using modified colours to make a menu-like overlay. Rather a more graphical visualization of static text, where some parts of the text are different depending on a variable.

knorth55, 2021-01-23:
it seems you already know how to deal with the issue, so can you make a PR?
we have not enough time to support all the feature request.
we can check and merge it.

xqrkushnir, 2021-01-23:
@knorth55 alright! Will be happy to do it. Could you please tell me how can I test my code? :blush: I can't figure out where to put freshly modified source files to see the change in RViz. My RViz was installed with `sudo apt install ros-melodic-desktop-full`, if that helps.

knorth55, 2021-01-23:
@xqrkushnir actually we don't have test code for these visualization tools, just building and several small tests.
If you make a PR, we can test it on our machine and merge it.

closed 2021-02-04 by xqrkushnir