jsk-ros-pkg/jsk_visualization #809: chainer などはどうインストールしていますか？

k-okada, 2021-04-13:
@708yamaguchi によると，chianer==6.7.0 -> cupy-cuda101==6.7.0 となっているけど，cudaの10.1に依存しているけど，普通に入るcudaは11などなので，インストールがむづかしいことになっている。

どうしていますか？ @knorth55

knorth55, 2021-04-13:
`sudo pip install cupy` すればどのCUDAでもインストール時にビルドされます

knorth55, 2021-04-13:
cupyはおそらくビルドできると思います
cupy-cuda111とかってないんですかね

708yamaguchi, 2021-04-13:
ありがとうございます、理解できてきました。
以下、自分メモです。

- `pip install cupy-cudaxxx`
  cudaxxxに対応しているビルド済のcupyをインストールする
- `pip install cupy`
  自分のPCにインストールされているcudaバージョンに対応するように、cupyのソースをダウンロードして自前でビルドする

前者は楽な一方、インストールしたいcupyのバージョンによっては対応しているcudaのバージョンに限りがある
後者はビルドに時間がかかって動作の保証もないが、好きなcudaのバージョンに合わせたcupyがインストールできる

python2.xではchainer==6.7.0を使う必要があるが、これに対応しているcupy==6.7.0を使おうとすると、cupy-cudaは101までしか対応していない。
最近のcudaは11.0以上が入っていることも多く、このようなcudaに対応するcupyをインストールしようとすると、後者の自前ビルドしか方法がない。