jsk-ros-pkg/jsk_visualization #819: RVIZ display are not loaded properly

maximilianwulf, 2021-08-04:
Hi, 

when trying to enable any RVIZ display I get the following error: 

```
[ERROR] [1628066690.062988630]: PluginlibFactory: The plugin for class 'jsk_rviz_plugin/AmbientSound' failed to load.  Error: Failed to load library /opt/ros/noetic/lib//libjsk_rviz_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libpcl_visualization.so.1.10: cannot open shared object file: No such file or directory)
```

I install the plugins through Debian on noetic.

maximilianwulf, 2021-08-04:
Hi @knorth55, yes: 

```
libpcl-dev  1.10.0+dfsg-6ubuntu1                    amd64        Point Cloud Library - development files
```

knorth55, 2021-08-04:
can you find libpcl_visualization.so in your computer?

maximilianwulf, 2021-08-04:
Yes, that is also available: 

```
libpcl-visualization1.10:amd64 1.10.0+dfsg-6ubuntu1 amd64 Point Cloud Library - visualization library
```

knorth55, 2021-08-05:
I tried with `osrf/ros:noetic-desktop-full` docker and installed `ros-noetic-jsk-rviz-plugins`, but I could not reproduce your error.
can you reproduce your error in dockerfile?

in my docker file, `libpcl_visualization` is found.

```
root@41198293fbc2:/# find / -name libpcl_visualization* 
/usr/lib/x86_64-linux-gnu/libpcl_visualization.so
/usr/lib/x86_64-linux-gnu/libpcl_visualization.so.1.10.0
/usr/lib/x86_64-linux-gnu/libpcl_visualization.so.1.10
root@41198293fbc2:/# dpkg -l | grep pcl
ii  libdapclient6v5:amd64                       3.20.5-1                          amd64        Client library for the Network Data Access Protocol
ii  libpcl-apps1.10:amd64                       1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - apps library
ii  libpcl-common1.10:amd64                     1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - common library
ii  libpcl-dev                                  1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - development files
ii  libpcl-features1.10:amd64                   1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - features library
ii  libpcl-filters1.10:amd64                    1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - filters library
ii  libpcl-io1.10:amd64                         1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - I/O library
ii  libpcl-kdtree1.10:amd64                     1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - kdtree library
ii  libpcl-keypoints1.10:amd64                  1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - keypoints library
ii  libpcl-ml1.10:amd64                         1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - ml library
ii  libpcl-octree1.10:amd64                     1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - octree library
ii  libpcl-outofcore1.10:amd64                  1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - outofcore library
ii  libpcl-people1.10:amd64                     1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - people library
ii  libpcl-recognition1.10:amd64                1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - recognition library
ii  libpcl-registration1.10:amd64               1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - registration library
ii  libpcl-sample-consensus1.10:amd64           1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - sample consensus library
ii  libpcl-search1.10:amd64                     1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - search library
ii  libpcl-segmentation1.10:amd64               1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - segmentation library
ii  libpcl-stereo1.10:amd64                     1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - stereo library
ii  libpcl-surface1.10:amd64                    1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - surface library
ii  libpcl-tracking1.10:amd64                   1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - tracking library
ii  libpcl-visualization1.10:amd64              1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - visualization library
ii  ros-noetic-jsk-recognition-msgs             1.2.15-1focal.20210423.235949     amd64        ROS messages for jsk_pcl_ros and jsk_perception.
ii  ros-noetic-pcl-conversions                  1.7.2-1focal.20210424.000016      amd64        Provides conversions from PCL data types and ROS message types
ii  ros-noetic-pcl-msgs                         0.3.0-1focal.20210423.235755      amd64        Package containing PCL (Point Cloud Library)-related ROS messages.
ii  ros-noetic-pcl-ros                          1.7.2-1focal.20210424.000906      amd64        PCL (Point Cloud Library) ROS interface stack.
ii  ros-noetic-perception-pcl                   1.7.2-1focal.20210424.003621      amd64        PCL (Point Cloud Library) ROS interface stack.
root@41198293fbc2:/# dpkg -l | grep jsk
ii  ros-noetic-jsk-footstep-msgs                4.3.2-1focal.20210423.225200      amd64        jsk_footstep_msgs
ii  ros-noetic-jsk-gui-msgs                     4.3.2-1focal.20210423.233852      amd64        jsk_gui_msgs
ii  ros-noetic-jsk-hark-msgs                    4.3.2-1focal.20210423.223738      amd64        jsk_hark_msgs
ii  ros-noetic-jsk-recognition-msgs             1.2.15-1focal.20210423.235949     amd64        ROS messages for jsk_pcl_ros and jsk_perception.
ii  ros-noetic-jsk-recognition-utils            1.2.15-1focal.20210716.093001     amd64        C++ library about sensor model, geometrical modeling and perception.
ii  ros-noetic-jsk-rviz-plugins                 2.1.7-4focal.20210716.094858      amd64        The jsk_rviz_plugins package
ii  ros-noetic-jsk-topic-tools                  2.2.11-2focal.20210424.001314     amd64        jsk_topic_tools
```

maximilianwulf, 2021-08-05:
Thank you very much for your answer, on my system `libpcl_visualization` is missing: 

```
[noetic/lib] ldd libjsk_rviz_plugins.so | grep pcl
	libpcl_search.so.1.10 => /usr/lib/x86_64-linux-gnu/libpcl_search.so.1.10 (0x00007f23e34d9000)
	libpcl_segmentation.so.1.10 => /usr/lib/x86_64-linux-gnu/libpcl_segmentation.so.1.10 (0x00007f23e275f000)
	libpcl_surface.so.1.10 => /usr/lib/x86_64-linux-gnu/libpcl_surface.so.1.10 (0x00007f23e19db000)
	libpcl_common.so.1.10 => /usr/lib/x86_64-linux-gnu/libpcl_common.so.1.10 (0x00007f23e1925000)
	libpcl_visualization.so.1.10 => not found
	libpcl_octree.so.1.10 => /usr/lib/x86_64-linux-gnu/libpcl_octree.so.1.10 (0x00007f23dd5d7000)
	libpcl_kdtree.so.1.10 => /usr/lib/x86_64-linux-gnu/libpcl_kdtree.so.1.10 (0x00007f23dd47e000)
	libpcl_ml.so.1.10 => /usr/lib/x86_64-linux-gnu/libpcl_ml.so.1.10 (0x00007f23dd45a000)
	libpcl_features.so.1.10 => /usr/lib/x86_64-linux-gnu/libpcl_features.so.1.10 (0x00007f23dace0000)
	libpcl_filters.so.1.10 => /usr/lib/x86_64-linux-gnu/libpcl_filters.so.1.10 (0x00007f23da4c4000)
	libpcl_sample_consensus.so.1.10 => /usr/lib/x86_64-linux-gnu/libpcl_sample_consensus.so.1.10 (0x00007f23d97d2000)
	libdapclient.so.6 => /usr/lib/x86_64-linux-gnu/libdapclient.so.6 (0x00007f23cf1f7000)
```

maximilianwulf, 2021-08-05:
I installed the package manually from `https://ubuntu.pkgs.org/20.04/ubuntu-universe-amd64/libpcl-visualization1.10_1.10.0+dfsg-5ubuntu1_amd64.deb.html` with 

`sudo dpkg -i libpcl-visualization1.10_1.10.0+dfsg-5ubuntu1_amd64.deb`

which solved the issue. Thank you very much for the help, much appreciated.

closed 2021-08-05 by maximilianwulf

yulongheart, 2024-05-15:
Hello, I have encountered some issues. It seems that there is an undefined symbol, "_ZN4rviz12EnumPropertyC1ERK7QStringS3_S3-PNS_8PropertyE", when loading the libjsk_rviz_plugins.so dynamic link library. I would like to ask if you have encountered this issue and how to solve it.
[image]