jsk-ros-pkg/jsk_visualization #821: CameraInfo | use image topic

DasSchilli, 2021-08-06:
Hello,
thank you for these ingenious plugins. I have a question regarding the CameraInfo DisplayType. As soon as I select an image of type sensor_msgs/Image (image::encodingsMONO16), the program crashes. Would it be possible to support more formats except RGB or is this not possible?

knorth55, 2021-08-06:
yes it is possible, but we do not have enough resource.
we want you to change the following codes and make a PR if possible.
https://github.com/jsk-ros-pkg/jsk_visualization/blob/master/jsk_rviz_plugins/src/camera_info_display.cpp

DasSchilli, 2021-08-11:
ok, now I can select an image with an image::encodings::TYPE_16UC1, but it crashes because of the methods drawImageTexture(). Sorry I am new to openCV and cannot understand what happens exactly in this method.

knorth55, 2021-08-11:
`image_` is considered as `RGB8`, so you need to change cv_ptr from `16UC1` to `RGB8` before https://github.com/jsk-ros-pkg/jsk_visualization/blob/master/jsk_rviz_plugins/src/camera_info_display.cpp#L411.

how you can do it is first change it from `16UC1` to `8UC1`, and then change it to `RGB8`.
something like these.

```c++
cv::Mat gray_image;
cv::normalize(cv_ptr->image, gray_image, cv::NORM_MINMAX, 0, 255);
image_ = cv::cvtColor(gray_image, cv::COLOR_GRAY2RGB);
```