jsk-ros-pkg/jsk_visualization #828: Error:  The plugin for class 'jsk_rviz_plugin/BoundingBoxArray' failed to load.

WiNet-Lab-IIT-Hyderabad, 2021-09-21:
BoundingBoxArray
The class required for this display, 'jsk_rviz_plugin/BoundingBoxArray', could not be loaded.
Error:
Failed to load library /opt/ros/melodic/lib//libjsk_rviz_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/ros/melodic/lib/libjsk_recognition_utils.so: undefined symbol: _ZNK3pcl6search6SearchINS_8PointXYZEE7getNameB5cxx11Ev)

k-okada, 2021-09-22:
Thank you for report. Could you try `roslaunch jsk_rviz_plugins boundingbox_sample.launch`? If this not working, may be there are some mismatch on library versions. Try `sudo apt update` and `sudo apt dist-upgrade` 
c.f. https://answers.ros.org/question/194341/rviz-motion_planning-error-failed-to-load-library/, https://answers.ros.org/question/200885/could-not-load-library-after-latest-ros-updates/

WiNet-Lab-IIT-Hyderabad, 2021-09-22:
> Thank you for report. Could you try `roslaunch jsk_rviz_plugins boundingbox_sample.launch`? If this not working, may be there are some mismatch on library versions. Try `sudo apt update` and `sudo [...]

Still getting same error:
 roslaunch jsk_rviz_plugins boundingbox_sample.launch
... logging to /home/bhaskar/.ros/log/6abb0d24-1aaf-11ec-b327-ac1f6b874df7/roslaunch-bhaskar-12313.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://bhaskar:40249/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    bounding_box_sample (jsk_rviz_plugins/bounding_box_sample.py)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[bounding_box_sample-1]: started with pid [12347]
process[rviz-2]: started with pid [12348]
Traceback (most recent call last):
  File "/opt/ros/melodic/share/jsk_rviz_plugins/samples/bounding_box_sample.py", line 7, in <module>
    from tf.transformations import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf/__init__.py", line 30, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)
[bounding_box_sample-1] process has died [pid 12347, exit code 1, cmd /opt/ros/melodic/share/jsk_rviz_plugins/samples/bounding_box_sample.py __name:=bounding_box_sample __log:=/home/bhaskar/.ros/log/6abb0d24-1aaf-11ec-b327-ac1f6b874df7/bounding_box_sample-1.log].
log file: /home/bhaskar/.ros/log/6abb0d24-1aaf-11ec-b327-ac1f6b874df7/bounding_box_sample-1*.log
[ERROR] [1632289535.857127810]: PluginlibFactory: The plugin for class 'jsk_rviz_plugin/BoundingBoxArray' failed to load.  Error: Failed to load library /opt/ros/melodic/lib//libjsk_rviz_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/ros/melodic/lib/libjsk_recognition_utils.so: undefined symbol: _ZNK3pcl6search6SearchINS_8PointXYZEE7getNameB5cxx11Ev)
[rviz-2] process has finished cleanly
log file: /home/bhaskar/.ros/log/6abb0d24-1aaf-11ec-b327-ac1f6b874df7/rviz-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

sktometometo, 2021-09-22:
@WiNet-Lab-IIT-Hyderabad 
```
Traceback (most recent call last):
File "/opt/ros/melodic/share/jsk_rviz_plugins/samples/bounding_box_sample.py", line 7, in
from tf.transformations import *
File "/opt/ros/melodic/lib/python2.7/dist-packages/tf/init.py", line 30, in
from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/init.py", line 38, in
from tf2_py import *
File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_py/init.py", line 38, in
from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)
[bounding_box_sample-1] process has died [pid 12347, exit code 1, cmd
```
This looks like that python3 node try to load python2 version of tf2_ros. Maybe you used other python binary other than system python with pyenv or somthing like that. Pleaes check your python environment.

Surtr07, 2022-11-05:
do you solve it?I meet same problem.

siddharthaZPdubey, 2023-03-03:
i am getting this error 
Errors     << point_cloud:make /home/siddhartha/zmr_sim_ws/logs/point_cloud/build.make.117.log                       
/home/siddhartha/zmr_sim_ws/src/point_cloud/src/point_cloud.cpp:7:10: fatal error: jsk_rviz_plugins/BoundingBoxArray.h: No such file or directory
    7 | #include <jsk_rviz_plugins/BoundingBoxArray.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/point_cloud.dir/build.make:63: CMakeFiles/point_cloud.dir/src/point_cloud.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:423: CMakeFiles/point_cloud.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

knorth55, 2023-03-03:
@siddharthaZPdubey your error is not same error and different from this issue.
Please open different issue.