jsk-ros-pkg/jsk_visualization #834: Error transforming pose from frame 'lidar' to frame 'map'

ajinkyakhoche, 2021-11-03:
I'm extracting a BoundingBoxArray() by running a 3d object detector on a point cloud. The boxes are visualized correctly when the rviz frame is set to 'lidar' (i.e. local frame). But when i select the global/fixed frame (called 'map'), rviz gives error and I don't see anything visualized. Why? The TF tree is correct (i.e. transform between 'map' and 'lidar' exists).  Any help would be appreciated.

k-okada, 2021-11-04:
@ajinkyakhoche thanks for report, could you provide more information on how to reproduce your problem?
ROS package with launch file, rviz config file, and node to publish frames are fine.

ajinkyakhoche, 2021-11-04:
Please find link to folder [jsk_visualization_issue_834](https://kth-my.sharepoint.com/:f:/g/personal/khoche_ug_kth_se/Ek8nuxqhDKZChdgCNcRFFx0Bq6Kf243RUj377jBCx5IX8Q?e=J6drgV)

Let me know if you can't access it. I've uploaded a rosbag with lidar point cloud (/lidar_top), bounding boxes (/pp_boxes) and /tf. I've also uploaded the rviz conf and a pdf showing tf frames.

Problem occurs when i change fixed frame from lidar_top to map

k-okada, 2021-11-05:
@ajinkyakhoche thank you for nice example. I think the root cause is inconsistent time stamp between `header` and `boxes` . The time in  header and tf seems consistent, but boxes is not. Could you check your node to publish the bounding boxes? How do they add timestamp?
```
---
header: 
  seq: 104
  stamp: 
    secs: 1532402931
    nsecs: 348383000
  frame_id: "lidar_top"
boxes: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1636050718
        nsecs: 956121683
      frame_id: "lidar_top"
    pose: 
      position: 
        x: 7.88828659058
        y: 16.3989486694
        z: -0.3765630126
      orientation: 
```
One solution is to modify messages
```
#!/usr/bin/env python

import rospy
from jsk_recognition_msgs.msg import BoundingBoxArray

def cb(msg):
    for b in msg.boxes:
        print(b)
        b.header.stamp = msg.header.stamp
        #b.header.stamp = rospy.Time(0)
    pub.publish(msg)

rospy.init_node('fix_bbox')
pub = rospy.Publisher("pp_bboxes_fix", BoundingBoxArray)
sub = rospy.Subscriber("pp_boxes", BoundingBoxArray, cb)
rospy.spin()
```
Subscribing `pp_bboxes_fix` and change Fixed frame to `map` seems ok.

Another solution is to use timestamp in `msg->header.stamp` when call `transform()` in pulugin, see #836.

closed 2021-11-08 by ajinkyakhoche