jsk-ros-pkg/jsk_visualization #837: Resource not found: jaxon_ros_bridge  (Maybe about: trans_system repository?)

devemin, 2021-11-16:
Dear jsk,

こんにちは。日本語でひとつご報告を、失礼致します。

先日、こちらのページにある
「moveit_msgs/DisplayRobotStateを利用してrviz上にロボットを表示させる」
https://github.com/jsk-ros-pkg/jsk_visualization/blob/master/doc/jsk_interactive_marker/display_robot_state.md
を試してみようと思ったのですが、

>>Resource not found: jaxon_ros_bridge
となり、詰まってしまいました。

どうやら jaxon_ros_bridge は、 trans_system リポジトリが使えない事に起因する？かもしれないとのことで、

私はROS 初心者で、DisplayRobotState を単に一度試してみたいだけの状況ではありましたが、
お手すきの際に、最初のページに jaxon_ros_bridge が使えない旨の記載か、jaxon_ros_bridge 公開などにより、先のページの実行環境構築が可能になるようであればご対応頂ければ幸いです。

一応のご報告でしたので、お手数おかけしてすみません。
宜しくお願い致します。

参考：インストール試行錯誤の様子
https://twitter.com/devemin/status/1460395397659037697?s=20

Thanks,

devemin

sktometometo, 2021-11-17:
https://github.com/jsk-ros-pkg/jsk_visualization/pull/838
このPull RequestでPR2で動くようになるかと思います。

このサンプルで動かしているスクリプトは、JSKでロボットを動かす際に使用しているEuslispでロボットの行動をプランニングした際に、プランニング結果をRViz上に、実機の現在の状態とは関係なく可視化するためのものです。
@devemin さんはご自身のプランニングソフトウェアを開発されていらっしゃると思うので、ご自身のソフトウェアから動作させる前のプランニング結果の確認として `moveit_msgs/DisplayRobotState` をpublishされるような使い方になると思いますので、こちらのデモは一つの `moveit_msgs/DisplayRobotState` の使用例とお考えいただければ幸いです。

devemin, 2021-11-17:
@sktometometo 様
サンプル、とても助かります！速い…！
moveit_msgs/DisplayRobotState についても理解しました。
帰宅後、試してみますね。本当にありがとうございます！

closed 2021-11-18 by devemin