jsk-ros-pkg/jsk_visualization #839: change colur of PieCart with code

OmerKonan, 2021-11-24:
Hello, I want to use piechart as warning info and I want to change colur based on input value. Is there a way to change the color useing my code?

closed 2021-11-24 by OmerKonan

shmpwk, 2021-11-24:
The easiest way is to change the piechart color on rviz manually and save the config. 
e.g. According to the  [piechart doc](https://jsk-visualization.readthedocs.io/en/latest/jsk_rviz_plugins/plugins/pie_chart.html?highlight=piechart), we have sample code. 
`roslaunch jsk_rviz_plugins overlay_sample.launch` shows below by default.

[image]

You can manually select the color at foreground color. 

[image]

shmpwk, 2021-11-24:
How about enabling `auto color change`?

OmerKonan, 2021-11-24:
I found it but I can not set threshold to change color. Also I have 3 different situation so `auto color change` enabling helps but not exactly solves my problem. Thank you

shmpwk, 2021-11-25:
Like this ? https://github.com/jsk-ros-pkg/jsk_visualization/pull/840

OmerKonan, 2021-11-25:
> 

Yes, That's perfect solution. Thank you so much!