jsk-ros-pkg/jsk_visualization #843: New release

Wimll, 2022-01-10:
Hi!

Are there any plans to release a new debian package? I am currently using the jsk visualization package, but right now I have to compile it from source on every machine because I need this fix for the visualization of filled ploygons:

https://github.com/jsk-ros-pkg/jsk_visualization/pull/820

As mentioned in this issue:

https://github.com/jsk-ros-pkg/jsk_visualization/issues/799

But as this fix was merged on 15th of September 2021 and the latest available tag (2.1.7) is from 17 Oct 2020, this fix is not in the debian package installed with apt get.

Are there any plans to release a new version in the future?

Thanks!

k-okada, 2022-01-11:
released on melodic and noetic https://github.com/ros/rosdistro/pull/31727, https://github.com/ros/rosdistro/pull/31728

closed 2022-01-11 by k-okada

Wimll, 2022-01-11:
@k-okada , thanks! I'll wait for the release to get merged, this saves me a lot of time (: