jsk-ros-pkg/jsk_visualization #847: How to install jsk_rviz_plugins?

wanglijia666, 2022-04-18:
Hello, I have downloaded zip through GitHub: https://github.com/jsk-ros-pkg/jsk_visualization, so how can I install  jsk_rviz_plugins in the ROS environment of Ubuntu? thank you.

knorth55, 2022-04-18:
please try `deb` package first.
```
apt-get install ros-noetic-jsk-rviz-plugins
# or
apt-get install ros-melodic-jsk-rviz-plugins
```

wanglijia666, 2022-04-18:
I have installed through apt-get, but failed. Can I download zip locally and install it again?

wanglijia666, 2022-04-18:
> please try `deb` package first.
> 
> ```
> apt-get install ros-noetic-jsk-rviz-plugins
> # or
> apt-get install ros-melodic-jsk-rviz-plugins
> ```

I have installed through apt-get, but failed. Can I download zip locally and install it again?

wanglijia666, 2022-04-18:
I have installed by apt-get, but failed. Can I download zip locally and install it again?

wanglijia666, 2022-04-18:
apt-get install ros-kinetic-jsk-rviz-plugins

k-okada, 2022-04-19:
@wanglijia666 do you use ubuntu 16.04?

wanglijia666, 2022-04-19:
yes，ubuntu 16.04

knorth55, 2022-04-19:
>I have installed by apt-get, but failed.

What do you mean `failed`?
Can you share us the error log?

wanglijia666, 2022-04-19:
Thank you very much. I solved this problem by changing the source.

k-okada, 2022-04-20:
That’s good news . Could you share more detail about the solution. What do
you mean by the ‘source’ and how did you change it ?

wanglijia666, 2022-04-20:
It may be because the download speed of using the default source is too slow, which leads to unsuccessful Download all the time. After switching the source, it is loaded successfully. sudo apt-get update.

closed 2023-05-10 by mqcmd196