jsk-ros-pkg/jsk_visualization #854: shift keyboard release issue in jsk_picker_tool ，The KeyRelease inside is not responding

lidongxu111, 2022-10-13:
int PickerTool::processKeyEvent(QKeyEvent* event, rviz::RenderPanel* panel)
{ 
  // bool processed = false;
  if (event->type() == QEvent::KeyPress && event->key() ==  Qt::Key_Shift ) { // shift event->type() == QEvent::KeyPress
    // if (event->isAutoRepeat()){
        // event->ignore();
        shift_pressing_ = true;
        // processed == true;
  }
  else if (event->type() == QEvent::KeyRelease && event->key() == Qt::Key_Shift)  {
    shift_pressing_ = false;
    // event->accept();
  }
  cout << "shift_pressing: " << shift_pressing_ <<endl;
  return 0;
}

k-okada, 2022-10-13:
Thank you for report, could you create PR for this?

lidongxu111, 2022-10-13:
I don't know how to create PR, but I can send the complete code to your email, OK？

lidongxu111, 2022-10-13:
Because my knowledge of the code was borrowed from the code

lidongxu111, 2022-10-14:
So, can you help me？