jsk-ros-pkg/jsk_visualization #858: How to offline install this package?

Charlie-lusie, 2022-10-18:
As title said, I want to use jsk vis on a offline x86 ubuntu 20.04 machine. I tried to directly compile the source code, but error occured. Can any one tell me how to offline install jsk vis?

knorth55, 2022-10-18:
we can build this package without internet.
please paste the whole build output.
we want to know the error.

Charlie-lusie, 2022-10-18:
> we can build this package without internet. please paste the whole build output. we want to know the error.

I put the package into a ros workspace folder like this:
ws_jsk/
         src/
                jsk_visualizatoion_msater/
                        jsk_inter.../
                        jsk_rqt/
                        jsk_rviz/
                        jsk_visualization/
                other_package/
the output is:
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake: 83(find_package):
    Could not find a package configuration file provided by "jsk_hark_msgs"
     with any of the following names:
            jsk_hark_msgsConfig.cmake
            jsk_hark_msgs-config.cmake
Add the installation prefix of "jsk_hark_msgs" to CMAKE_PREFIX_PATH or set "jsk_hark_msgs_DIR" ...

knorth55, 2022-10-18:
please install the dependencies
```
cd ws_jsk
rosdep install --from-path src -y -r -i
```

knorth55, 2022-10-18:
you need jsk_common_msgs and pther dependencies.
i dont know how you download the source but you can download deb file from ros server and manually install it, or download source and build it

closed 2023-05-10 by mqcmd196