jsk-ros-pkg/jsk_visualization #859: How to write multiple rows of text with OverlayText using CPP?

Freddd13, 2022-10-22:
Thank you for the awesome work!
From the python sample script in this repo, I managed to display an one-row text displayed on RVIZ.
However, when I try to display multiple rows of text, there's nothing displayed, and nothing got when use `rostopic echo {TEXT topic}
```c++
// this not works:
  text.text = "this is the first row\n this is the second row";
// either did this:
  text.text = "this is the first row\
this is the second row";
```

In the python example, you just use `"""` to write just as you wish to display, but it seems not to work with cpp. 
So I wonder  how to display multiple rows of text using cpp? Any help is appreciated!

mqcmd196, 2023-05-07:
It's weird, could you paste full cpp code?

Freddd13, 2023-05-08:
```c++
  jsk_rviz_plugins::OverlayText text;
  text.width = width;
  text.height = height;
  text.left = left;
  text.top = top;
  text.text_size = fontsize;
  text.line_width = line_width;
  text.text = content;

  std_msgs::ColorRGBA color1, color2;
  color1.r = fr;
  color1.g = fg;
  color1.b = fb;
  color1.a = fa;

  color2.r = br;
  color2.g = bg;
  color2.b = bb;
  color2.a = ba;

  text.fg_color = color1;
  text.bg_color = color2;
  text+= "t_whole: " + std::to_string(XXXXXX) + " ms"; // here's the text

  puber.publish(text);
```
In fact, I managed to get the desired display by adding  a <br> at the end of the text.
I was inspired by a QT related blog and this method really worked. But still wondering what's the best practice?

mqcmd196, 2023-05-09:
I have no problem with the code below

```cpp
#include <jsk_rviz_plugins/OverlayText.h>
#include <ros/ros.h>

int main(int argc, char **argv){
    ros::init(argc, argv, "overlay_text_publisher");
    ros::NodeHandle nh;
    ros::Publisher pub = ros::NodeHandle().advertise<jsk_rviz_plugins::OverlayText>("overlay_text", 1);

    jsk_rviz_plugins::OverlayText text;
    text.width = 400;
    text.height = 600;
    text.left = 10;
    text.top = 10;
    text.text_size = 12;
    text.line_width = 2;
    text.font = "DejaVu Sans Mono";
    text.text = "this is the first row\n this is the second row";

    while(ros::ok()){
        pub.publish(text);
        ros::spinOnce();
    }
}
```

[image]

I think
> However, when I try to display multiple rows of text, there's nothing displayed, and nothing got when use `rostopic echo {TEXT topic}

is weird. Did you check the publisher work correctly?

Freddd13, 2023-05-09:
Yes, that's why I use <br> instead. I was using Ubuntu 18 under wsl. Maybe it's something with the environment?

Sent with a Spark

mqcmd196, 2023-05-10:
I think adding `<br>` is nothing to do with not seeing rostopic by using `rostopic echo`. 

>  I was using Ubuntu 18 under wsl. Maybe it's something with the environment?

I suspect this point. Some network configurations might not work properly. 

BTW, I tested my code on Ubuntu 20.04 on wsl2 and worked correctly... anyway please reopen this issue if you have trouble.

closed 2023-05-10 by mqcmd196