jsk-ros-pkg/jsk_visualization #872: About 'OverlayCamera' display

vislero, 2023-04-26:
When I try to use the 'overlay camera' display in rviz, it only shows a black rectangle in the render panel.

On the other hand, in rviz's original 'camera' display, it works fine on the same topic data.

So I analyzed your code and it seems to work if I uncomment the  below code it works.

https://github.com/jsk-ros-pkg/jsk_visualization/blob/aa401204c985af679ffe02e4af121a05240cd675/jsk_rviz_plugins/src/overlay_camera_display.cpp#L244

But at the same time, there is a problem that another identical display is created in a new window.

mqcmd196, 2023-05-07:
I haven't encountered

> When I try to use the 'overlay camera' display in rviz, it only shows a black rectangle in the render panel.

Could you share the rosbag file and rviz settings that reproduce this problem? Or if rviz outputs something error, please share it

vislero, 2023-05-15:
OK the url below contains the rviz config file and bag file that i recorded.

[link](https://uvify-my.sharepoint.com/:u:/p/daehyun_hwang/EUOkmfBLVShFmd1UL1jFKKoBCktA6PVewRykd_hRoMs9eg?e=ABzZYa)

i test with Relasense d-455 to check this issue. 

thank you.

mqcmd196, 2023-05-16:
[image]

It works well in my environment ( Ubuntu 20.04, ros noetic, using the `master` branch ) without configuration editing. Did you build 
any other related packages from the source?

vislero, 2023-05-16:
> 
> It works well in my environment ( Ubuntu 20.04, ros noetic, using the `master` branch ) without configuration editing. Did you build any other related packages from the source?

No I just installed the library 

using below command 

'sudo apt install ros-noetic-jsk-rviz-plugins'