jsk-ros-pkg/jsk_visualization #887: symbol lookup error while adding jsk_rviz_plugins in rviz

Danendra10, 2024-05-21:
At first, everything works fine, but all of sudden my rviz force close, and i can't add the bounding box array... does anyone know what the issue is :(

I'm using ubuntu 20, ros noetic

[image]

Here is the rospack

```
ackermann_steering_controller
actionlib
actionlib_msgs
actionlib_tutorials
amcl
angles
apriltag_ros
async_web_server_cpp
audio_common_msgs
autoware_msgs
base_local_planner
bond
bondcpp
bondpy
camera_calibration
camera_calibration_parsers
camera_info_manager
can_msgs
catkin
class_loader
clear_costmap_recovery
cmake_modules
compressed_depth_image_transport
compressed_image_transport
control_msgs
control_toolbox
controller_interface
controller_manager
controller_manager_msgs
costmap_2d
cpp_common
cv_bridge
ddynamic_reconfigure
depth_image_proc
depthimage_to_laserscan
diagnostic_aggregator
diagnostic_analysis
diagnostic_common_diagnostics
diagnostic_msgs
diagnostic_updater
diff_drive_controller
dwa_local_planner
dynamic_reconfigure
dynamic_tf_publisher
easy_markers
effort_controllers
eigen_conversions
face_detector
filters
find_object_2d
force_torque_sensor_controller
forward_command_controller
gazebo_dev
gazebo_msgs
gazebo_plugins
gazebo_ros
gazebo_ros_control
gencpp
geneus
genlisp
genmsg
gennodejs
genpy
geodesy
geographic_msgs
geometry_msgs
gl_dependency
gmapping
grid_map_core
grid_map_cv
grid_map_msgs
grid_map_ros
gripper_action_controller
hardware_interface
hector_gazebo_plugins
hector_gazebo_thermal_camera
hector_gazebo_worlds
hector_mapping
hls_lfcd_lds_driver
husky_navigation
image_geometry
image_proc
image_publisher
image_rotate
image_transport
image_view
image_view2
imu_filter_madgwick
imu_sensor_controller
interactive_marker_tutorials
interactive_markers
joint_limits_interface
joint_state_controller
joint_state_publisher
joint_state_publisher_gui
joint_trajectory_controller
jsk_footstep_msgs
jsk_gui_msgs
jsk_hark_msgs
jsk_interactive
jsk_interactive_marker
jsk_interactive_test
jsk_recognition_msgs
jsk_recognition_utils
jsk_rqt_plugins
jsk_rviz_plugins
jsk_topic_tools
kalman_filter
kdl_conversions
kdl_parser
laser_assembler
laser_filters
laser_geometry
leg_detector
lgsvl_msgs
libg2o
libnabo
libpointmatcher
librviz_tutorial
map_laser
map_msgs
map_server
media_export
message_filters
message_generation
message_runtime
mk
move_base
move_base_msgs
moveit_msgs
nav_core
nav_msgs
navfn
nmea_msgs
nodelet
nodelet_topic_tools
nodelet_tutorial_math
object_recognition_msgs
octomap
octomap_msgs
openslam_gmapping
pcl_conversions
pcl_msgs
pcl_ros
people_msgs
people_tracking_filter
people_velocity_tracker
pluginlib
pluginlib_tutorials
pointcloud_to_laserscan
polled_camera
posedetection_msgs
position_controllers
python_qt_binding
qt_dotgraph
qt_gui
qt_gui_cpp
qt_gui_py_common
qwt_dependency
realsense2_camera
realsense2_description
realtime_tools
resource_retriever
robot_localization
robot_state_publisher
ros_environment
rosapi
rosauth
rosbag
rosbag_migration_rule
rosbag_storage
rosbash
rosboost_cfg
rosbridge_library
rosbridge_msgs
rosbridge_server
rosbuild
rosclean
rosconsole
rosconsole_bridge
roscpp
roscpp_serialization
roscpp_traits
roscpp_tutorials
roscreate
rosgraph
rosgraph_msgs
roslang
roslaunch
roslib
roslint
roslisp
roslz4
rosmake
rosmaster
rosmsg
rosnode
rosout
rospack
rosparam
rosparam_shortcuts
rospy
rospy_tutorials
rosserial_client
rosserial_msgs
rosserial_python
rosservice
rostest
rostime
rostopic
rosunit
roswtf
roswww
rotate_recovery
rqt_action
rqt_bag
rqt_bag_plugins
rqt_console
rqt_dep
rqt_graph
rqt_gui
rqt_gui_cpp
rqt_gui_py
rqt_image_view
rqt_launch
rqt_logger_level
rqt_moveit
rqt_msg
rqt_nav_view
rqt_plot
rqt_pose_view
rqt_publisher
rqt_py_common
rqt_py_console
rqt_reconfigure
rqt_robot_dashboard
rqt_robot_monitor
rqt_robot_steering
rqt_runtime_monitor
rqt_rviz
rqt_service_caller
rqt_shell
rqt_srv
rqt_tf_tree
rqt_top
rqt_topic
rqt_web
rtabmap
rtabmap_conversions
rtabmap_costmap_plugins
rtabmap_demos
rtabmap_examples
rtabmap_launch
rtabmap_legacy
rtabmap_msgs
rtabmap_odom
rtabmap_python
rtabmap_ros
rtabmap_rviz_plugins
rtabmap_slam
rtabmap_sync
rtabmap_util
rtabmap_viz
rviz
rviz_plugin_tutorials
rviz_python_tutorial
self_test
sensor_msgs
shape_msgs
smach
smach_msgs
smach_ros
smclib
sound_play
stage
stage_ros
std_msgs
std_srvs
stereo_image_proc
stereo_msgs
teleop_twist_keyboard
tf
tf2
tf2_eigen
tf2_geometry_msgs
tf2_kdl
tf2_msgs
tf2_py
tf2_ros
tf2_sensor_msgs
tf2_tools
tf2_web_republisher
tf_conversions
theora_image_transport
topic_tools
trajectory_msgs
transmission_interface
turtle_actionlib
turtle_tf
turtle_tf2
turtlebot3_bringup
turtlebot3_description
turtlebot3_fake
turtlebot3_gazebo
turtlebot3_msgs
turtlebot3_navigation
turtlebot3_teleop
turtlesim
unique_id
urdf
urdf_parser_plugin
urdf_sim_tutorial
urdf_tutorial
urdfdom_py
uuid_msgs
velocity_controllers
velodyne_description
velodyne_driver
velodyne_gazebo_plugins
velodyne_laserscan
velodyne_msgs
velodyne_pointcloud
view_controller_msgs
visualization_marker_tutorials
visualization_msgs
voxel_grid
web_video_server
webkit_dependency
xacro
xmlrpcpp
```

mqcmd196, 2024-05-22:
I can't understand what happened, only the information you provided.

Symbol lookup errors are commonly resolved by running `sudo apt update && sudo apt upgrade` and rebuilding your whole workspace packages.

If you don't fix it, please paste the result of 
```shell
dpkg -l | grep ros-noetic-
```

and 

```shell
vcs status
```

under your `<catkin workspace>/src`

mqcmd196, 2024-05-25:
@Danendra10 Could you execute 
```shell
sudo apt update && sudo apt upgrade
```
and build your workspace? I think the version incompatibility with rviz and jsk_visualization is the problem.  CC: @shin0805

Danendra10, 2024-05-26:
Thanks for your assist, it turns out all of the problem in my ros are started because I installed the python3-rosdep2, it made all of my ros noetic deps crash, so I just reinstalled the ROS and everything work fine. Thank you, i'll close this Issue

closed 2024-05-26 by Danendra10