jsk-ros-pkg/jsk_visualization #900: Failed to load jsk_rviz_plugin/BoundingBoxArray due to undefined symbol in libpcl_io.so.1.10

yooyo-Q, 2025-08-20:
Environment:
- ROS Distribution: Noetic
- OS: [Please specify your operating system, e.g., Ubuntu 20.04 LTS]
- jsk-rviz-plugins version: [2.1.10-1focal.20250521.014225]

Description:
I'm encountering an error when trying to use the jsk_rviz_plugin/BoundingBoxArray in ROS Noetic. After installing the jsk plugins, the plugin fails to load with a library loading error related to an undefined symbol in libpcl_io.so.1.10.

Error Message:
[ERROR] [1755670356.368276335]: PluginlibFactory: The plugin for class 'jsk_rviz_plugin/BoundingBoxArray' failed to load.  Error: Failed to load library /opt/ros/noetic/lib//libjsk_rviz_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /lib/x86_64-linux-gnu/libpcl_io.so.1.10: undefined symbol: libusb_set_option)

Steps to Reproduce:
1. Install jsk-rviz-plugins on ROS Noetic
2. Launch rviz with a configuration that includes jsk_rviz_plugin/BoundingBoxArray
3. Observe the error in the console output

Expected Behavior:
The jsk_rviz_plugin/BoundingBoxArray plugin should load successfully without errors.

Actual Behavior:
Plugin fails to load with the error message mentioned above, citing an undefined symbol libusb_set_option in libpcl_io.so.1.10.

[image]

yooyo-Q, 2025-08-21:
This is my environment in ROS(noetic) docker environment

`(base) root@yk-GeekPro-17IRR:/home/yk# echo $CMAKE_PREFIX_PATH
/opt/ros/noetic
(base) root@yk-GeekPro-17IRR:/home/yk# rospack find rviz
/opt/ros/noetic/share/rviz
(base) root@yk-GeekPro-17IRR:/home/yk# rospack find jsk_rviz_plugins
/opt/ros/noetic/share/jsk_rviz_plugins
(base) root@yk-GeekPro-17IRR:/home/yk# dpkg -l | grep rviz
ii  ros-noetic-jsk-rviz-plugins                 2.1.10-1focal.20250521.014225     amd64        The jsk_rviz_plugins package
ii  ros-noetic-librviz-tutorial                 0.11.0-1focal.20220212.163831     amd64        Tutorial showing how to compile your own C++ program with RViz displays and features.
ii  ros-noetic-rqt-rviz                         0.7.0-1focal.20220328.224826      amd64        rqt_rviz provides a GUI plugin embedding RViz.
ii  ros-noetic-rviz                             1.14.14-1focal.20220212.160156    amd64        3D visualization tool for ROS.
ii  ros-noetic-rviz-plugin-tutorials            0.11.0-1focal.20220212.164049     amd64        Tutorials showing how to write plugins for RViz.
ii  ros-noetic-rviz-python-tutorial             0.11.0-1focal.20220212.164051     amd64        Tutorials showing how to call into rviz internals from python scripts.
ii  ros-noetic-visualization-msgs               1.13.1-1focal.20210423.224023     amd64        visualization_msgs is a set of messages used by higher level packages, such as rviz, that deal in visualization-specific data.
ii  ros-noetic-visualization-tutorials          0.11.0-1focal.20220212.164542     amd64        Metapackage referencing tutorials related to rviz and visualization.
(base) root@yk-GeekPro-17IRR:/home/yk# dpkg -l | grep pcl
ii  libdapclient6v5:amd64                       3.20.5-1                          amd64        Client library for the Network Data Access Protocol
ii  libpcl-apps1.10:amd64                       1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - apps library
ii  libpcl-common1.10:amd64                     1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - common library
ii  libpcl-dev                                  1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - development files
ii  libpcl-features1.10:amd64                   1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - features library
ii  libpcl-filters1.10:amd64                    1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - filters library
ii  libpcl-io1.10:amd64                         1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - I/O library
ii  libpcl-kdtree1.10:amd64                     1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - kdtree library
ii  libpcl-keypoints1.10:amd64                  1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - keypoints library
ii  libpcl-ml1.10:amd64                         1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - ml library
ii  libpcl-octree1.10:amd64                     1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - octree library
ii  libpcl-outofcore1.10:amd64                  1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - outofcore library
ii  libpcl-people1.10:amd64                     1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - people library
ii  libpcl-recognition1.10:amd64                1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - recognition library
ii  libpcl-registration1.10:amd64               1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - registration library
ii  libpcl-sample-consensus1.10:amd64           1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - sample consensus library
ii  libpcl-search1.10:amd64                     1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - search library
ii  libpcl-segmentation1.10:amd64               1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - segmentation library
ii  libpcl-stereo1.10:amd64                     1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - stereo library
ii  libpcl-surface1.10:amd64                    1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - surface library
ii  libpcl-tracking1.10:amd64                   1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - tracking library
ii  libpcl-visualization1.10:amd64              1.10.0+dfsg-5ubuntu1              amd64        Point Cloud Library - visualization library
ii  ros-noetic-jsk-recognition-msgs             1.2.19-1focal.20250520.012924     amd64        ROS messages for jsk_pcl_ros and jsk_perception.
ii  ros-noetic-pcl-conversions                  1.7.4-1focal.20250520.005606      amd64        Provides conversions from PCL data types and ROS message types
ii  ros-noetic-pcl-msgs                         0.3.0-1focal.20220107.005305      amd64        Package containing PCL (Point Cloud Library)-related ROS messages.
ii  ros-noetic-pcl-ros                          1.7.4-1focal.20250521.004323      amd64        PCL (Point Cloud Library) ROS interface stack.
ii  ros-noetic-perception-pcl                   1.7.4-1focal.20220221.112136      amd64        PCL (Point Cloud Library) ROS interface stack.
`

[... cut ...]
ii  ros-noetic-jsk-recognition-msgs             1.2.19-1focal.20250520.012924        amd64        ROS messages for jsk_pcl_ros and jsk_perception.
ii  ros-noetic-pcl-conversions                  1.7.4-1focal.20250520.005606         amd64        Provides conversions from PCL data types and ROS message types
ii  ros-noetic-pcl-msgs                         0.3.0-1focal.20250520.005335         amd64        Package containing PCL (Point Cloud Library)-related ROS messages.
ii  ros-noetic-pcl-ros                          1.7.4-1focal.20250521.004323         amd64        PCL (Point Cloud Library) ROS interface stack.
`

yooyo-Q, 2025-08-21:
I have observed that my host environment is consistent with yours. The difference is that Miniconda is installed in my system.

mqcmd196, 2025-08-21:
It seems that the jsk-rviz-plugins version is the same as mine, but rviz version is older than mine. Please upgrade rviz by `sudo apt upgrade` . The jsk-rviz-plugins sometimes breaks ABI compatibility. ( https://github.com/jsk-ros-pkg/jsk_visualization/issues/890 ) 

> I have observed that my host environment is consistent with yours. The difference is that Miniconda is installed in my system.

If the upgrading doesn't solve your issue, please deactivate conda. Mixing ROS environments and conda/venv environments is a very advanced technique.

yooyo-Q, 2025-08-29:
The error occurs due to a library conflict between HKRobotics MVS SDK and PCL (Point Cloud Library) related to `libusb`.

Hikvision Robotics MVS installs its own version of the `libusb` library, which is likely incompatible with the system's `libusb` required by PCL. When ROS/jsk_rviz_plugins tries to load `libpcl_io.so`, it's attempting to use the Hikvision Robotics MVS-provided `libusb` that lacks the `libusb_set_option` symbol required by the system's PCL installation.

This creates a symbol resolution failure because the MVS library path is prioritized in `LD_LIBRARY_PATH`, causing the wrong version of `libusb` to be loaded for PCL-dependent components.